zephyrproject-rtos/zephyr #38623: add support ZigBee stack for EFR32 MCU

cvetaevvitaliy, 2021-09-17:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

Please add support ZigBee for EFR32 chips

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Zephyr is a great OS, I would like to have ZigBee support for EFR32 chips

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

chrta, 2021-09-17:
This is related to  https://github.com/zephyrproject-rtos/zephyr/issues/17005

nashif, 2026-10-30:
Zigbee stack for EFR32 is vendor/out-of-tree material; no Zephyr Zigbee stack in tree.

closed 2026-10-30 by nashif