zephyrproject-rtos/zephyr #38624: mcuboot gets the wrong value of DT_FIXED_PARTITION_ID

maweicoltsmart, 2021-09-17:
project：zephyrproject/bootloader/mcuboot/samples/zephyr
branch：main
mcuboot commit：70bfbd21cdf5f6d1402bc8d0031e197222ed2ec0
zephyr commit：81595ed4db2081e227d6425acb25b6eec8191787
board & cpu：A microcontroller based on ARM Cortex-m0 designed in China
The dts file is modified as follows：
```
&flash0 {
	status = "okay";
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;
		status = "okay";
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x000000000 0x00010000>;
			read-only;
			status = "okay";
		};
		slot0_partition: partition@10000 {
			label = "image-0";
			reg = <0x000010000 0x000016400>;
			status = "okay";
		};
		slot1_partition: partition@26400 {
			label = "image-1";
			reg = <0x00026400 0x000016400>;
			status = "okay";
		};
		scratch_partition: partition@3c800 {
			label = "image-scratch";
			reg = <0x0003c800 0x2000>;
			status = "okay";
		};
		storage_partition: partition@3e800 {
			label = "storage";
			reg = <0x0003e800 0x00000800>;
			status = "okay";
		};
	};
};
```
The kconfig file is modified as follows：
```
CONFIG_FLASH=y
CONFIG_IMG_MANAGER=y
CONFIG_FLASH_PAGE_LAYOUT=y
```
Compile and burn and run
`zephyrproject/bootloader/mcuboot/samples/zephyr$ make BOARD=wch_eth_ch579 full.bin`

After running, the value of flash_map[i].fa_id output by the code in the file zephyrproject/zephyr/subsys/storage/flash_map/flash_map.c is respectively: 32, 43, 66, 43, 104

```
static struct flash_area const *get_flash_area_from_id(int idx)
{
	printk("%s, %d, %d, %d\r\n", __func__, __LINE__, idx, flash_map_entries);
	for (int i = 0; i < flash_map_entries; i++) {
		if (flash_map[i].fa_id == idx) {
			return &flash_map[i];
		}else {
			printk("%s, %d, %d, %d\r\n", __func__, __LINE__, idx, flash_map[i].fa_id);
		}
	}

	return NULL;
}
```

de-nordic, 2021-09-21:
The `fa_id` is taken from generated file devicetree_unfixed.h and depends on number of defined partitions and which partitions area enabled.
Do you happen to have DTS with lots of "fixed-partitions' compatible nodes, with many of them disabled?
The "issue" has been described here: https://github.com/zephyrproject-rtos/zephyr/issues/36517

maweicoltsmart, 2021-09-23:
The zephyr.dts partition definition in the build directory is like this：
```
		flash: flash-controller@40001800 {
			compatible = "wch,ch57x-flash-controller";
			label = "FLASH_CTRL";
			reg = < 0x40001800 0x200 >;
			#address-cells = < 0x1 >;
			#size-cells = < 0x1 >;
			flash0: flash@0 {
				compatible = "soc-nv-flash";
				label = "FLASH_WCH";
				reg = < 0x0 0x3f000 >;
				erase-block-size = < 0x200 >;
				write-block-size = < 0x4 >;
				status = "okay";
				partitions {
					compatible = "fixed-partitions";
					#address-cells = < 0x1 >;
					#size-cells = < 0x1 >;
					status = "okay";
					boot_partition: partition@0 {
						label = "mcuboot";
						reg = < 0x0 0x10000 >;
						read-only;
						status = "okay";
					};
					slot0_partition: partition@10000 {
						label = "image-0";
						reg = < 0x10000 0x16400 >;
						status = "okay";
					};
					slot1_partition: partition@26400 {
						label = "image-1";
						reg = < 0x26400 0x16400 >;
						status = "okay";
					};
					scratch_partition: partition@3c800 {
						label = "image-scratch";
						reg = < 0x3c800 0x2000 >;
						status = "okay";
					};
					storage_partition: partition@3e800 {
						label = "storage";
						reg = < 0x3e800 0x800 >;
						status = "okay";
					};
				};
			};
		};
```

de-nordic, 2021-09-27:
@maweicoltsmart Can you share list of PARTITION_ID from your devicetree header file?
```
grep PARTITION_ID zephyr/include/generated/devicetree_unfixed.h
```

maweicoltsmart, 2021-09-28:
@
```
grep PARTITION_ID build/wch_eth_ch579/mcuboot/zephyr/include/generated/devicetree_unfixed.h

#define DT_N_S_soc_S_flash_controller_40001800_S_flash_0_S_partitions_S_partition_0_PARTITION_ID 0
#define DT_N_S_soc_S_flash_controller_40001800_S_flash_0_S_partitions_S_partition_10000_PARTITION_ID 1
#define DT_N_S_soc_S_flash_controller_40001800_S_flash_0_S_partitions_S_partition_26400_PARTITION_ID 2
#define DT_N_S_soc_S_flash_controller_40001800_S_flash_0_S_partitions_S_partition_3c800_PARTITION_ID 3
#define DT_N_S_soc_S_flash_controller_40001800_S_flash_0_S_partitions_S_partition_3e800_PARTITION_ID 4
```
But the function in the file flash_map.c: 
```
static struct flash_area const *get_flash_area_from_id(int idx)
{
	printk("%s, %d, %d\r\n", __func__, __LINE__, flash_map_entries);
	for (int i = 0; i < flash_map_entries; i++) {
		if (flash_map[i].fa_id == idx) {
			return &flash_map[i];
		}else{
			printk("%s, %d, %d, %d\r\n", __func__, __LINE__, flash_map[i].fa_id, idx);
		}
	}

	return NULL;
}
```

The obtained flash_map_entries value is 5, and the flash_map[i].fa_id values are 32, 43, 66, 43, 104 respectively

de-nordic, 2021-09-30:
@maweicoltsmart Do you use `FLASH_AREA_ID` to obtain area ID, for example `FLASH_AREA_ID(storage)` ?

maweicoltsmart, 2021-10-02:
@de-nordic
The value of flash_map[i].fa_id comes from the initialization in the file zephyr/subsys/storage/flash_map/flash_map_default.c:
```
#define FLASH_AREA_FOO(part)						\
	{.fa_id = DT_FIXED_PARTITION_ID(part),				\
	 .fa_off = DT_REG_ADDR(part),					\
	 .fa_dev_name = DT_LABEL(DT_MTD_FROM_FIXED_PARTITION(part)),	\
	 .fa_size = DT_REG_SIZE(part),},
```

de-nordic, 2021-10-04:
Yes, I know where it comes from. The `DT_FIXED_PARTITION_ID` that is used to initialize the `fa_id` uses DT macros to get to the `..._PARTITION_ID` for each known partition, this is the only way.
The `flash_map_entries` may be different from the last ID because it counts number of entries that are in the array; the ID is generated for each partition but only enabled partitions (status=okay) are included. This means that if you have for example seven partitions and the first two are disabled, then the `flash_map_entries` will be five, and IDs does not have to be neither continuous nor ordered.

For the  `DT_FIXED_PARITION_ID` to get strange values that  do not match these from the `devicetree_unfixed.h` does not seem to be possible.

carlescufi, 2021-11-09:
Unable to reproduce, please reopen if still an issue.

closed 2021-11-09 by carlescufi