zephyrproject-rtos/zephyr #38629: Power Electronics Roadmap

martinjaeger, 2021-09-17:
This issue is used to track the status of features related to power electronics in Zephyr.

See also related [Mini-Conference recording](https://www.youtube.com/watch?v=9XKQqPuZh9w&ab_channel=ZephyrProject) from Zephyr Developer Summit 2021.

## Advanced PWM / timers

### Typical features

- Complementary PWM channels
- Automatic dead-time insertion
- Synchronization between multiple channels (typically at least 3)
- Tight coupling with other peripherals

### Overview of MCU series

MCU Series    | Motor Timer                                  | High-resolution Timer
--------------|----------------------------------------------|-----------------------
ST STM32      | TIM1/TIM8 - Advanced Timer                   | HRTIM - High-Resolution Timer
Microchip SAM | TCC - Timer/Counter for Control Applications | -
NXP Kinetis   | FlexTimer                                    | eFlexPWM
TI C2000 *    | ePWM | HRPWM

*) currently not supported by Zephyr

### ToDos (mainly for STM32 HRTIM)

- [x] Support of pinmuxing for HRTIM via Devicetree (https://github.com/zephyrproject-rtos/hal_stm32/pull/87)
- [ ] Support of HRTIM in existing PWM driver (single channel, but high resolution)
- [ ] Create new `half_bridge` / `sync_pwm` / `complementary_pwm` API (see also [RIOT implementation](https://github.com/RIOT-OS/RIOT/pull/15165)).
- [ ] Implement new API for STM32 HRTIM

## Real-Time Stream and Sensor API (RTIO)

Offloading of ADC and automated processing of ADC readings in the background.

### Related PRs

- https://github.com/zephyrproject-rtos/zephyr/pull/17434

## Hardware triggers

Hardware triggers to connect timers, ADC, DAC and comparators.

### Related PRs

- https://github.com/zephyrproject-rtos/zephyr/pull/33918

## Comparator API

Mainly relevant if it can be coupled with DAC and timers for fast overcurrent protection and DC/DC control loops.

henrikbrixandersen, 2021-09-17:
I have implemented support for the NXP Kinetis Analog Comparator (ACMP) using the Zephyr sensors API: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/mcux_acmp/mcux_acmp.c

Perhaps we can generalize the sensor API extensions used for the ACMP and avoid a comparator specific API?

hongshui3000, 2022-11-18:
Is this roadmap still a work in progress?I feel like I've been stuck for a lot of time.

seyednasermoravej, 2024-02-22:
Hello everyone,

I’m working on implementing a converter that consists of a few cascaded devices. Each of these devices requires a high-resolution timer, and it’s essential that these timers are synchronized with each other. The system-on-chip (SoC) I’m using is the STM32G474RE, which includes a high-resolution timer. However, I couldn’t find any information about this timer in the .dtsi file. My question is whether it’s possible to utilize this timer and achieve synchronization among the timers. In CubeMX, the input synchronization pin is referred to as HRTIM1_SCIN, and the output synchronization is labeled as HRTIM1_SCOUT.