zephyrproject-rtos/zephyr #38631: printk to console fails for freescale kinetis 8.2.0 (Zephyr 2.6.0) on FRDM-K64F

torge2, 2021-09-17:
**Problem Description**
Printk outputs are not sent to the console using Freescale Kinetis 8.2.0, Zephyr 2.6.0, PlatformIO Core 5.20 (Home 3.3.4). Platform is FRDM-K64F. 

**To Reproduce**
1. Install most recent PlatformIO v.2.3.3 in VS Studio Code (Core 5.2.0, Home 3.3.4).
2. Install Platfrom "Freescale Kinetis", ver. 8.2.0
3. Try a blinky App and additionally add one line of code at the beginning of main:
_printk("hello world! %s\n", CONFIG_BOARD);_
4. prj.conf looks like
CONFIG_GPIO=y
CONFIG_SERIAL=y
5. build and upload to FRDM-K64F

**Expected behavior**
Green LED should start blinking, "hello world! ..." should be seen in console .
**Received behavior**
The green LED is blinking only if CONFIG_SERIAL=n. LED is not blinking if CONFIG_SERIAL=y, printk statements are sent to console in no case.

**Workaround**
Using the FRDM-K64F controller with the same simple application (blinky + "HelloWorld") is working with Freescale Kinetis 8.1.0, Zephyr 2.5.0. LED is blinking properly, printk statements are sent to the console.

**Impact**
Without logging statements to the console application development is more complicated.

**console output**
No output.

**Environment**
 - OS: Windows 10
 - PlatformIO v.2.3.3 (Core 5.2.0), Zephyr 2.6.0, Freescale Kinetis 8.2.0
 - FRDM-K64F

dleach02, 2021-09-20:
@torge2, you need file this bug with the PlatformIO project.

torge2, 2021-09-20:
So is this working with Zephyr 2.6.0 without PlatformIO? For me this is not clear.

dleach02, 2021-10-10:
@tux19 there are two primary methods for adjusting initialization order. The runlevel can be changed as you are asking or the priority within the runlevel. The SIM and UART are currently initialized both in PRE_KERNEL_1 with a priority of CONFIG_KERNEL_INIT_PRIORITY_DEVICE. So both using the same level/priority which means their real order of initialization is undefined and as you are seeing in this bug report, the linker can change the order they are stored in the device array. Clearly there is a dependency between the UART and the clock SIM since the UART init function calls uart_mcux_configure() which then tries to get the clock object as you point out.

I think the simple fix is to change the priority the UART is initialized so that it initializes after the SIM. I'll look to see how best to do this.

dleach02, 2021-10-10:
@tux19, can you try my PR #39308 to ensure it addresses your problem?

closed 2021-10-17 by nashif