zephyrproject-rtos/zephyr #38653: drivers: modem: gsm_ppp: Add support for Quectel modems

ycsin, 2021-09-19:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Multiple Quetel modems supports the PPP and GSM MUX, the currentl `gsm_ppp` driver should be able to support these modems with minimal changes.

**Describe why you are asking for this support?**
I'm currently using Quectel modems in my custom board.

**Additional context**
The [CMUX document](https://forums.quectel.com/uploads/short-url/uoB16C44M5IzrdgmZ87PPJ9axSc.pdf) is available [publicly on the internet](https://www.google.com/search?q=quectel+at%2Bcmux+site%3Aforums.quectel.com&rlz=1C1SQJL_enMY906MY907&oq=quectel+at%2Bcmux+site%3Aforums.quectel.com).
Based on my testing, a Quectel modem can support the GSM MUX without publicly stating it in the reference manual, so Quectel user should just try.

tcpipchip, 2021-09-19:
Hi,
I am using that driver, GSM_PPP (Generic), but i never getting the GATEWAY and  MASK, they always are returning 0.0.0.0 and 0.0.0.0
I only got the IP
I am testing in three u-blox models, LISA U200, G350 and G450.
Looks that the problem is that we dont have the USER and PASSWORD option on GENERIC_MODEM

ycsin, 2021-09-20:
@tcpipchip 

> Looks that the problem is that we dont have the USER and PASSWORD option on GENERIC_MODEM
> Yes, the GSM_PPP needs a reset option to MODEM :(

Maybe create another issue(s) for these?
This issue is for the support of Quectel modems, and will be closed automatically when PR #38652 get merged

tcpipchip, 2021-09-20:
Ok!

PR #38652 (2021-09-23): drivers: modem: gsm_ppp: Add support for Quectel modems w/ CMUX
Add support for Quectel modem in `gsm_ppp` modem driver.

The CMUX cmd is based on MUX application notes v1.0(2020-09-22) for BG95, BG77 and BG600L. Tested and working on EC21e (EC21EFAR06A01M4G or newer).

As the `gsm_ppp` doesn't do power control, it is required to power on the modem manually in the application.

The [CMUX document](https://forums.quectel.com/uploads/short-url/uoB16C44M5IzrdgmZ87PPJ9axSc.pdf) is available [publicly on the internet](https://www.google.com/search?q=quectel+at%2Bcmux+site%3Aforums.quectel.com&rlz=1 [... cut]
closed 2021-09-23 by nashif