zephyrproject-rtos/zephyr #38654: drivers: modem: bg9x: Has no means to update size of received packet.

ycsin, 2021-09-19:
**Describe the bug**
This issue is to document a unresolved bug that I encountered when I first tried the ZephyrRTOS with the `bg9x` modem driver on my EC21 modm.

The current `quectel-bg9x` driver's recv isn't working as expected, as the [recv's unsolicited cmd handler](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/quectel-bg9x.c#L378) doesn't update the packet size for the socket, compared to the [ublox-sara-r4 driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/ublox-sara-r4.c#L920) as a reference.

This is related to #34273, where the MQTT `CONNACK` acknowledgement from the broker can never be read by the offloaded socket's implementation, since the data to be read is always zero.

In earlier version of Quectel modems or its firmware, when the modem receives something, the modem reports the `<connectID>` without specifying the size. For the current `quectel-bg9x` to work on these modem, it is required to query the size of the received buffer using a separate AT cmd, likely in a work.

Some recent firmware (> 2020) for some of the modems have been updated to report the `length` together with the `<connectID>` if configured to do so. For these lucky modems with the correct firmware, it is possible to update the buffer size directly in the [unsol recv parser](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/quectel-bg9x.c#L378) just like the [ublox-sara's one](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/ublox-sara-r4.c#L920).

**To Test**
1. Build mqtt samples on a board using the bg9x driver
2. See error

**Impact**
`bg9x` driver will not be able to receive anything.

**Additional context**
I've tried to fix the issue last time, IIRC, MQTT QoS 0 works after the patch but QoS 1 likely fails. I have since move to use `gsm_ppp` modem.

The commit can be found [here](https://github.com/ycsin/zephyr/commit/34a778eee5b447c03495a268e98b6373e206734d) for whoever interested to pick up this, but it is very messy as it contains some CCLK and GNSS stuff. I came up with this with the help of @elliot-wdtl 

Some important things:
1. The old [`on_cmd_unsol_recv`](https://github.com/ycsin/zephyr/blob/prev_develop/drivers/modem/quectel-bg9x.c#L486) is modified to submit a work to perform [`modem_qird_query_work`](https://github.com/ycsin/zephyr/blob/prev_develop/drivers/modem/quectel-bg9x.c#L441)
2. [`modem_qird_query_work`](https://github.com/ycsin/zephyr/blob/prev_develop/drivers/modem/quectel-bg9x.c#L441) will do the [AT cmd](https://github.com/ycsin/zephyr/blob/prev_develop/drivers/modem/quectel-bg9x.c#L459) to [query the size of the received buffer](https://github.com/ycsin/zephyr/blob/prev_develop/drivers/modem/quectel-bg9x.c#L414) and [update the socket](https://github.com/ycsin/zephyr/blob/prev_develop/drivers/modem/quectel-bg9x.c#L467) accordingly.

elliot-wdtl, 2021-09-19:
> Some recent firmware (> 2020) for some of the modems have been updated to report the length together with the <connectID> if configured to do so.

This is interesting info. I'm working with bg95 and I have made quite a few changes which I haven't got to trying to upstream yet. I take advantage of configuring the modem to indicate the length so maybe this will be a problem if it's not available on lots of modems. I have also found that even if I configure it to indicate the length, it will not indicate the length on SSL sockets which is a sticky point for me. You also can't query the size of the receive buffer on SSL sockets so I have a different workaround.

ycsin, 2021-09-21:
> I have made quite a few changes which I haven't got to trying to upstream yet. I take advantage of configuring the modem to indicate the length so maybe this will be a problem if it's not available on lots of modems.

I guess for older firmware's users they will have to implement the AT+QIRD way, at least your current implementation will cover lots of newer firmware and modems

closed 2022-08-30 by github-actions[bot]