zephyrproject-rtos/zephyr #38655: Failing Tests for Regulator API

leonardp, 2021-09-19:
**Describe the bug**
The following testcases are failing:
drivers.regulator.fixed.onoff.on_delay
drivers.regulator.fixed.onoff.off_delay

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/drivers/regulator/fixed
2. add `boards/<board>.overlay` if needed
3. `west build -b <board> . -p -- -DDTS_EXTRA_CPPFLAGS=-DAPP_DTS_REGULATOR_ON_DELAYED`
4. west flash
5. inspect output

**Expected behavior**
Succeeding testcase.

**Impact**
I want to make an API for adjustable regulators with very similar functionality to `regulator-fixed`.
With broken `_DELAYED` options the Regulator API is basically just a wrapper for toggling GPIOs through
"On-Off-Manager".

**Console output of the failing tests**

```
*** Booting Zephyr OS build v2.6.0-rc1-3858-g0d538447144c  ***
reg 0x8006620 gpio 0x80065a8
Regulator: regulator-fixed
startup-delay: 10000 us
off-on-delay: 0 us
Running test suite regulator_test
===================================================================
START - test_preconditions
 PASS - test_preconditions in 0.1 seconds
===================================================================
START - test_basic
D: test: start
D: test: finalize 0 delay 10000 us

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/regulator/fixed/src/main.c:226: test_basic: (rc not equal to -EAGAIN)
startup notify early: 0
D: test: work delay complete
D: test: finalize 0 delay 0 us
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/lib/os/onoff.c:283
E: r0/a1:  0x00000004  r1/a2:  0x0000011b  r2/a3:  0x80000000
E: r3/a4:  0x080029cd r12/ip:  0xa0000000 r14/lr:  0x08000d5d
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x08005dc8
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x200002f8 (sysworkq)
E: Halting system
```
```
*** Booting Zephyr OS build v2.6.0-rc1-3858-g0d538447144c  ***
reg 0x80065e8 gpio 0x8006570
Regulator: regulator-fixed
startup-delay: 0 us
off-on-delay: 20000 us
Running test suite regulator_test
===================================================================
START - test_preconditions
 PASS - test_preconditions in 0.1 seconds
===================================================================
START - test_basic
D: test: start
D: test: finalize 0 delay 0 us
D: test: stop
D: test: finalize 0 delay 20000 us
 PASS - test_basic in 0.9 seconds
===================================================================
Test suite regulator_test succeeded
===================================================================
PROJECT EXECUTION SUD: test: work delay complete
D: test: finalize 0 delay 0 us
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/lib/os/onoff.c:283
E: r0/a1:  0x00000004  r1/a2:  0x0000011b  r2/a3:  0x80000000
E: r3/a4:  0x08002995 r12/ip:  0xa0000000 r14/lr:  0x08000d25
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x08005d90
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x200002f8 (sysworkq)
E: Halting system

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnuarmemb (gcc-arm-embedded-10.3.1)
 - Commit SHA or Version used: 0d538447144cda0f15cde4c800efb353da541bc5

**Additional Context**

Example of a *succeeding* test:
```
west build -b blackpill_f411ce . -p && west flash

*** Booting Zephyr OS build v2.6.0-rc1-3858-g0d538447144c  ***
reg 0x80065e4 gpio 0x800656c
Regulator: regulator-fixed
startup-delay: 0 us
off-on-delay: 0 us
Running test suite regulator_test
===================================================================
START - test_preconditions
 PASS - test_preconditions in 0.1 seconds
===================================================================
START - test_basic
D: test: start
D: test: finalize 0 delay 0 us
D: test: stop
D: test: finalize 0 delay 0 us
 PASS - test_basic in 0.9 seconds
===================================================================
Test suite regulator_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```
with `boards/blackpill_f411ce.overlay`
```
#include "test_common.dtsi"

/ {
	regulator {
		enable-gpios = <&gpiob 4 GPIO_ACTIVE_HIGH>;
	};
	resources {
		check-gpios = <&gpiob 5 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>;
	};
};
```

jfischer-no, 2021-09-21:
I confirm the error on nrf52840dk.

erwango, 2021-09-22:
@cfriedt Assigning you back on this as I haven't found any codeowner or maintainer aside from @pabigot whom I think is not active nowadays on the project.

PR #38789 (2021-09-24): drivers: regulator: Fix k_work_schedule return code handling
k_work_schedule may return other non-negative value than 0.
When driver was adapted to the new k_work API that was not
taken into account.

Fixes #38655.
closed 2021-09-24 by nashif