zephyrproject-rtos/zephyr #38665: Add real support for MIMXRT1050_EVK_QSPI

ainguraXmarquiegui, 2021-09-19:
**Is your enhancement proposal related to a problem? Please describe.**
Zephyr currently has a compile target named `mimxrt1050_evk_qspi` that any newcomer to the platform would expect to work as straightforward as any other target. Unfortunately, images compiled for this target cannot be flashed with `west flash`, and if flashed with some other methods, they will not boot.

**Describe the solution you'd like**
I would like to integrate my proposed changes to both zephyr repository and `modules/hal/nxp`:

https://github.com/ainguraXmarquiegui/zephyr/tree/feature/mimxrt1050_evk_qspi_support
https://github.com/ainguraXmarquiegui/hal_nxp/tree/feature/mimxrt1050_evk_qspi_support

The hal changes adds support for the QSPI flash. My code change proposition uses the config flags to decide whether to compile hyperflash or QSPI flash support.

On the zephyr part, I propose to add a link to the SEGGER documentation where they explain what needs to be manually modified to be able to flash a QSPI board. When mergin this code changes, `westl.yml` shoud point to the correct hal revision with support for QSPI boot.

**Describe alternatives you've considered**
I think improving the documentation, and adding code changes to really support the platform are our best alternative. It would be even better that SEGGER would add support for both hyperflash and QSPI, but I think that's outside the scope of this project.

ainguraXmarquiegui, 2021-09-19:
@MaureenHelm
@dleach02

mmahadevan108, 2021-09-24:
Thank you for the changes, kindly submit your PR's and we will be happy to review and merge.

ainguraXmarquiegui, 2021-11-27:
We can close the issue and keep tracking progress in the pull requests.

closed 2021-11-27 by ainguraXmarquiegui