zephyrproject-rtos/zephyr #38667: ST LSM6DSO polling mode does not work on nRF52dk_nrf52832

kclauah, 2021-09-20:
**Describe the bug**
```
2021-09-20T02:07:51.541Z DEBUG modem << *** Booting Zephyr OS build v2.6.0-rc1-ncs1  ***
2021-09-20T02:08:05.338Z DEBUG modem << Testing LSM6DSO sensor in polling mode.
2021-09-20T02:08:05.360Z DEBUG modem << [00:00:00.accel x:0.000000 ms/2 y:0.000000 ms/2 z:0.000000 ms/2
2021-09-20T02:08:05.362Z DEBUG modem << 361,267] gyro x:0.000000 dps y:0.000000 dps z:0.000000 dps
2021-09-20T02:08:05.364Z DEBUG modem << trig_cnt:1
2021-09-20T02:08:05.367Z DEBUG modem << [0m<inf> LSM6DSO: chip id 0x6c[0m
2021-09-20T02:08:06.354Z DEBUG modem << accel x:0.000000 ms/2 y:0.000000 ms/2 z:0.000000 ms/2
2021-09-20T02:08:06.369Z DEBUG modem << gyro x:0.000000 dps y:0.000000 dps z:0.000000 dps
2021-09-20T02:08:06.370Z DEBUG modem << trig_cnt:2
2021-09-20T02:08:07.370Z DEBUG modem << accel x:0.000000 ms/2 y:0.000000 ms/2 z:0.000000 ms/2
2021-09-20T02:08:07.382Z DEBUG modem << gyro x:0.000000 dps y:0.000000 dps z:0.000000 dps
```
Running on nRF52dk_nrf52832. Also tried with sparkfun thing plus nrf9160.

**To Reproduce**

1. add nrf52dk_nrf52832.overlay in the project
```
arduino_i2c: &i2c0 {
	lsm6dso@6a {
		compatible = "st,lsm6dso";
		label = "LSM6DSO";
		reg = <0x6a>;
	};
};
```
2. add overlay in the CMakeLists.txt file
```
set(<image_name>_DTC_OVERLAY_FILE
  ${CMAKE_CURRENT_SOURCE_DIR}/nrf52dk_nrf52832.overlay
)
```
3. west build -b nrf52dk_nrf52832 zephyr\samples\sensor\lsm6dso
4. west flash
5. See 0

anangl, 2021-09-29:
What compatible is used for `i2c0`, is it "nordic,nrf-twi" or "nordic,nrf-twim", can you check in `zephyr.dts`?
Could you try to test the sample with the latest main, at least with this commit: 99ce264df772e9fdfd053896c5b7834547e169c2? It looks like the problem could be the same as the one reported in #33440 that should be already fixed by #38678.

kclauah, 2021-09-29:
> What compatible is used for i2c0, is it "nordic,nrf-twi" or "nordic,nrf-twim", can you check in zephyr.dts?
it is "nordic,nrf-twim";

> Could you try to test the sample with the latest main ...
It works for me with nrf52dk_nrf52832 and lsm6dso in polling mode. 

> It looks like the problem could be the same as the one reported in #33440 that should be already fixed by #38678.
It happens when I test it on nRF Connect SDK v1.6.1 which does not include this commit: 99ce264

anangl, 2021-09-29:
> ` It looks like the problem could be the same as the one reported in #33440 that should be already fixed by #38678.` It happens when I test it on nRF Connect SDK v1.6.1 which does not include this [...]

In this case, adding `zephyr,concat-buf-size` in your overlay should help.

kclauah, 2021-09-30:
Thank you for your kind assistance.

closed 2021-09-30 by kclauah