zephyrproject-rtos/zephyr #38668: ESP32‘s I2S

sam131208, 2021-09-20:
Zephyr shall support I2S devices. ESP32 has been in zephyr for a long time, it is time to improve their driver support.

   ESP32's I2S integrates many other functions and is highly technical. It is more appropriate to be completed by zephyr.

Rhvarrier, 2023-02-04:
I see that this issue has been closed, but I still don't see any support for i2s for esp32.

ahj40, 2023-06-01:
I have to agree with Rhvarrier, I can't see any I2S support for ESP32 platforms. Is it possible that someone could be a bit more informative? Is it because it is not an 'issue' it is simply missing espressif drivers?

kartben, 2023-09-08:
@Rhvarrier @ahj40 see umbrella issue #29394 and recent comment https://github.com/zephyrproject-rtos/zephyr/issues/29394#issuecomment-1702244416