zephyrproject-rtos/zephyr #38675: DTS binding create devicetree_unfixed.h build error at v2.7.0

lgl88911, 2021-09-20:
**Describe the bug**
I added the dts binding file and devicetree node, it compiles normally under v2.6.0, but fails to compile under 2.7.0-rc2.

**To Reproduce**
Steps to reproduce the behavior:
1. Add file dts/bindings/sensor/rotary-encoder.yaml
```
description: |
  Sensor driver for the relative-axis rotary encoder

compatible: "rotary-encoder"

properties:
  label:
    type: string
    required: true
  a-gpios:
    type: phandle-array
    required: true
    description: A pin for the encoder
  b-gpios:
    type: phandle-array
    required: true
    description: B pin for the encoder
  ppr:
    type: int
    description: Pulse Per Revolution
    required: false
  spp:
    type: int
    description: |
        Number of steps (stable states) per period
        1: Full-period mode (default)
        2: Half-period mode
        4: Quarter-period mode
    required: false
```
2. add node to board devicetree
```
input_encoder: rotary_encoder {
        compatible = "rotary-encoder";
        status = "okay";
        label = "INPUT_ENCODER";
        a-gpios = <&gpio1 22 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
        b-gpios = <&gpio1 23 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
        ppr = <15>;
        spp = <2>;
};
```
3. Add driver file drivers/sensor/rotary_encoder/rotary_encoder.c
```
#define DT_DRV_COMPAT rotary_encoder

#include <device.h>
#include <drivers/gpio.h>
#include <drivers/sensor.h>
```
4. When I use west to build,See error
> /mnt/e/tomato/t-clock-master/build/zephyr/include/generated/devicetree_unfixed.h:723:39: error: expected ')' before string constant
  723 | #define DT_N_S_rotary_encoder_P_label "INPUT_ENCODER"

**Expected behavior**
Build Pass.

**Impact**
Can‘t migrate the Rotary encoder driver code from 2.6.0 to 2.7.0

**Additional context**
I check devicetree_unfixed.h:723:39， Everything seems to be normal, I don’t know why the compilation error occurs
```
#define DT_N_S_rotary_encoder_STATUS_okay 1

/* Pin control (pinctrl-<i>, pinctrl-names) properties: */
#define DT_N_S_rotary_encoder_PINCTRL_NUM 0

/* Generic property macros: */
#define DT_N_S_rotary_encoder_P_label "INPUT_ENCODER"
#define DT_N_S_rotary_encoder_P_label_STRING_TOKEN INPUT_ENCODER
#define DT_N_S_rotary_encoder_P_label_STRING_UPPER_TOKEN INPUT_ENCODER
```

mbolivar-nordic, 2021-09-27:
Hi, I can't reproduce this issue.

I tried to follow your steps to reproduce exactly using this branch:

https://github.com/mbolivar-nordic/zephyr/tree/bug-38675

And I tested this with:

```
west build -b nrf52840dk_nrf52840 samples/hello_world/ -p
```

The file drivers/sensor/rotary_encoder/rotary_encoder.c builds without error.

Can you please confirm this branch is what you are trying to do? If not, can you please push a commit based on that branch which reproduces the issue, so I can test it?

lgl88911, 2021-09-28:
@mbolivar-nordic Thanks for checking, Sorry,I didn’t write all the code, it wasted your extra checking time. I have found the cause of the problem, this is my mistake. In drivers/sensor/rotary_encoder/rotary_encoder.c, the code is as follows: 
```
#define DT_DRV_COMPAT rotary_encoder

#include <device.h>
#include <drivers/gpio.h>
#include <drivers/sensor.h>

//some driver code
....

#define ENCODER_INST(n)											      \
	struct encoder_data encoder_data_##n;								      \
	const struct encoder_config encoder_cfg_##n = {							      \
		.a_label = DT_INST_GPIO_LABEL(n, a_gpios),						      \
		.a_pin = DT_INST_GPIO_PIN(n, a_gpios),							      \
		.a_flags = DT_INST_GPIO_FLAGS(n, a_gpios),						      \
		.b_label = DT_INST_GPIO_LABEL(n, b_gpios),						      \
		.b_pin = DT_INST_GPIO_PIN(n, b_gpios),							      \
		.b_flags = DT_INST_GPIO_FLAGS(n, b_gpios),						      \
		COND_CODE_0(DT_INST_NODE_HAS_PROP(n, ppr), (1), (DT_INST_PROP(n, ppr))),		      \
		COND_CODE_0(DT_INST_NODE_HAS_PROP(n, spp), (SPP_FULL), (DT_INST_PROP(n, spp))),		      \
	};												      \
	DEVICE_AND_API_INIT(encoder_##n, DT_INST_LABEL(n), encoder_init, &encoder_data_##n, &encoder_cfg_##n, \
			    POST_KERNEL, CONFIG_SENSOR_INIT_PRIORITY, &encoder_driver_api);

DT_INST_FOREACH_STATUS_OKAY(ENCODER_INST)

```
DEVICE_AND_API_INIT is supported in 2.6.0 but has been marked as obsolete. 2.7.0 removed it, which cause build error at 2.7.0. 

Now I replaced DEVICE_AND_API_INIT with DEVICE_DT_INST_DEFINE, it can be compiled under 2.7.0.

closed 2021-09-28 by lgl88911