zephyrproject-rtos/zephyr #38683: stm32g0b1: spi: MOSI signal goes high before and after transmitting data

simonguinot, 2021-09-20:
I noticed this issue while using the ws2812_spi LED strip driver on a nucleo_g0b1re board.

The led_ws2812 sample can be used to reproduce the issue. This [patch](https://github.com/simonguinot/zephyr/commit/3a9b09aced5643bd30f7ea2404401f2928c773ac) is needed to add support for the nucleo_g0b1re board to the led_ws2812 sample.

When running this sample, then the MOSI signal goes high for 80 microseconds before and after transmitting data:

[image]

[image]

The issue is that this high "preamble" state is interpreted as a high bit by the first WS2812 LED of the strip. And then the signal of all the LEDs is shifted by one bit, which results in visual glitches.

Note that from a SPI protocol standpoint, the MOSI signal is correct because the clock signal is off during this extra "high" states. Thus, this won't affect a real SPI device. But the WS2812 strip LED controller is not a SPI device. It doesn't use the clock signal. SPI is used here as a hack on a single wire (MOSI) to generate a control signal for the LEDs.

In addition, the issue is not systematic. It depends on the last data sent over MOSI. If the last 4 bytes are zero, then there are no high "preamble" state when sending the next buffer of data.

I also tried disabling DMA and interrupts but this doesn't affect the issue.

simonguinot, 2021-09-20:
Hi @erwango. Maybe do you have some ideas about how to fix this behaviour ?

simonguinot, 2021-09-21:
Hi @FRASTM,

Thanks for testing.

> I can see the the application running on g071rb nucleo board.
> `west build -p auto -b nucleo_g071rb samples/drivers/led_ws2812/`
> setting default frame format will display R, G, B colors
> 
> ```
>  		spi-one-frame = <0x70>;
> 		spi-zero-frame = <0x40>;
> ```

OK. And do you have a high "preamble" or not on the MOSI signal ?

With this sample, you won't see many visual artifacts due to the extra bit.  So it may seem OK visually while it is not.

Please have a look at the signal.

If you need, I can provide a sample with a visual feedback when the bug is here.
 
> 
> With
> 
> ```
> #define B1414_ZERO_FRAME 0xC0
> #define B1414_ONE_FRAME   0xF0
> ```
> 
> --> white led

The ws2812_spi driver supports two different devices: WS2812 and B1414. They have different timings. The definition above is for B1414 devices and a 4MHz SPI frequency. So if you have a WS2812 device, or if you are using a different frequency, then this definition is not correct. If you need I can help to provide a correct definition for your configuration.

simonguinot, 2021-09-21:
> @simonguinot Do you have more info to characterize this bug (G0 only, Zephyr only, long preamble also present with other SPI devices .. ) ?

Hi @erwango.

Thanks for looking into this. Unfortunately I don't have much more information.

I can tell I never experimented this issue with F0 MCUs.

I don't have built a firmware using STM32Cube (because I don't have the setup), so I don't know if the issue is rather in Zephyr, or in the HAL/SDK, or in the SPI controller itself.

Also I tried to play with the pinmux configuration or the SPI mode, but it didn't work.
 
Because the last data transmitted affects the issue, I suspect it could be something related with the TXFIFO and the SPI initialization/transmission.

And that's pretty much all I have.

FRASTM, 2021-09-21:
If I keep the definition for the ws2812, the MOSI high level is seen during ~60µs  when CONFIG_SPI_STM32_DMA=y

FRASTM, 2021-09-24:
MOSI signal on nxp FRDM_k64f board when testing led samples/drivers/led_ws2812/
[image]

FRASTM, 2021-09-24:
@simonguinot, could try changing the SPI protocol for your environment **./drivers/spi/spi_ll_stm32.c**
```
@@ -559,7 +559,7 @@ static int spi_stm32_configure(const struct device *dev,
 
 #if !defined(CONFIG_SOC_SERIES_STM32F1X) \
 	&& (!defined(CONFIG_SOC_SERIES_STM32L1X) || defined(SPI_CR2_FRF))
-	LL_SPI_SetStandard(spi, LL_SPI_PROTOCOL_MOTOROLA);
+	LL_SPI_SetStandard(spi, LL_SPI_PROTOCOL_TI);
 #endif
```

FRASTM, 2021-09-27:
I do not know about different protocol "in Motorola mode it goes high and then low between bytes"
The [AN5543](https://www.st.com/content/ccc/resource/technical/document/application_note/group1/ad/cc/bd/4c/4a/74/4a/67/DM00725181/files/DM00725181.pdf/jcr:content/translations/en.DM00725181.pdf) explains :
_"standard SPI Motorola mode is not ideal on old SPI versions. The signal logic copies the SPE status and causes problems to handle its GPIO alternate function. When the SPI is disabled, the signal is no longer propagated onto the associated GPIOs causing its floating once the SPE bit is set to zero.
On recent SPI versions, control over the alternate function is maintained even if the SPI is disabled. Also it is
possible to select the polarity of the pin and to swap most of the MOSI and MISO signals capabilities. Recent SPI
versions only support single NSS/SS pin. This alternate function has to be replaced by standard GPIO(s) control
when handling a multi slave system or when a specific communication requires a reconfiguration of SPI within a
single session (such at Half-duplex mode when the SPI has to be disabled temporarily while SS needs to stay
active)._

avisconti, 2021-09-27:
I found this link here explaining the difference between the two protocols: 
https://itectec.com/electrical/electronic-spi-ti-or-motorola-mode/

It is related to CPOL and CPHA configuration. But I'm not sure to understand how this is related to this issue.
 
``` @@ -559,7 +559,7 @@ static int spi_stm32_configure(const struct device *dev,

#if !defined(CONFIG_SOC_SERIES_STM32F1X) \
 	&& (!defined(CONFIG_SOC_SERIES_STM32L1X) || defined(SPI_CR2_FRF))
-	LL_SPI_SetStandard(spi, LL_SPI_PROTOCOL_MOTOROLA);
+	LL_SPI_SetStandard(spi, LL_SPI_PROTOCOL_TI);
 #endif
```
My suggestion is to make this configuration available as DT option, because I guess that it might be different for SPI_1 or SPI_2 and so on.

FRASTM, 2021-09-28:
@avisconti, thank you

avisconti, 2021-09-28:
> @avisconti, thank you
> We could think of a ./dts/bindings/spist,stm32-spi-common.yaml with
> 
> ```
>     frame-format:
>       type: boolean
>       required: false
>       description: |
>         Use Frame Format to support TI or motorola SPI mode.
>         false: SPI Motorola mode
>         true: SPI TI mode
> ```

Yes, it seems good!

avisconti, 2021-09-28:
@frastm Maybe you need to specify also a 'default' case and put in the description the reaso for that choice?

erwango, 2021-09-28:
Updating the issue and removing the bug as it's not a bug, but how it's supposed to work.

erwango, 2021-09-28:
> @avisconti, thank you
> We could think of a ./dts/bindings/spist,stm32-spi-common.yaml with
> 
> ```
>     frame-format:
>       type: boolean
>       required: false
>       description: |
>         Use Frame Format to support TI or motorola SPI mode.
>         false: SPI Motorola mode
>         true: SPI TI mode
> ```

@tbursztyka do you think this should go in spi.yaml ?
Also, any suggestion on the name ?

erwango, 2021-10-01:
Reopening this as #38911 was reverted.
Assigned to @tbursztyka for a proper integration into SPI API

ottojo, 2021-11-05:
We ran into a similar issue on Teensy 4.1 (NXP imxrt1062), where the idle-state for MOSI seems to be high, with no option to change that. The strip pattern appeared shifted to the right by one LED, with glitches on the first LED (sk6812-2020). We worked around that by sending 0-bytes before and after the pattern of at least the specified reset delay: https://github.com/teamspatzenhirn/zephyr/pull/4

simonguinot, 2021-11-05:
If the same issues happens with different SPI controllers, then maybe it makes sense to have a workaround in the LED strip driver. But it can't be adding zeros to both ends of the buffer to transmit via SPI :) There are compatible LED devices that have a reset delay of 250 microseconds. With a 6MHz SPI frequency, two resets would cost 375 bytes on such devices... There must be a more elegant way to solve this. Once a SPI controller is initialized with the correct CPOL and CPHA configuration, then MOSI should be low when nothing is emitted. It should be able to ask the SPI driver for this state.

tbursztyka, 2021-11-11:
see #39990

simonguinot, 2021-11-11:
> see #39990

Thanks.

tbursztyka, 2021-12-02:
@FRASTM could you update the stm32 driver to take into account the newly introduced flag ? 

I need to ping all SPI drivers maintainers to do the same actually.

closed 2022-01-17 by nashif