zephyrproject-rtos/zephyr #38688: bt_gatt_unsubscribe does not remove subscription from internal list/returning BT_GATT_ITER_STOP causes bt_gatt_subscribe to return -ENOMEM / -12

pigeonhands, 2021-09-21:
**Describe the bug**
If BT_GATT_ITER_STOP is returned from the [`bt_gatt_subscribe_params::notify` callback](https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L2802), [`bt_gatt_unsubscribe` is called](https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L4554) which does not clear the internal state of the notification subscription correctly.

Trying to create new subscriptions to different devices with `bt_gatt_subscribe` will eventually start returning -ENOMEM (-12) when the internal subscription list fills up.

**To Reproduce**
Using the code below, call `start_subscription` on more devices than (`CONFIG_BT_MAX_PAIRED` + `CONFIG_BT_MAX_CONN`).
```C
static bt_gatt_subscribe_params sub_params = { 0 };

static uint8_t on_notify_data(struct bt_conn *conn, struct bt_gatt_subscribe_params *params, const void *data, uint16_t length){
    return BT_GATT_ITER_STOP;
}

void start_subscription(bt_conn * conn){
  sub_params.notify = on_notify_data;
  sub_params.value_handle = ctx.handles.measurement_value;
  sub_params.ccc_handle = ctx.handles.measurement_cccd;
  sub_params.value = BT_GATT_CCC_NOTIFY;
  atomic_set_bit(sub_params.flags, BT_GATT_SUBSCRIBE_FLAG_VOLATILE);
  int err = bt_gatt_subscribe(ctx.conn, &sub_params);
  if (err == -ENOMEM) {
      //bug has happened
  }
}
```

**Expected behavior**

New subscriptions should not fail. 

`bt_gatt_unsubscribe` should set the internal states `gatt_sub::peer` when the subscription list is empty like what `gatt_sub_remove` does:
https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L2715

**Impact**
In an environment where it is expected to enable subscriptions of many different peers over this devices lifetime, subscriptions will eventually fail and not allow any more subscriptions to be created.

**Environment (please complete the following information):**
NRF Sdk 1.6.1

**Additional context**
A temprary fix for this is to just to always return `BT_GATT_ITER_CONTINUE` from the `bt_gatt_subscribe_params::notify` callback and not call `bt_gatt_unsubscribe` anywhere.

joerchan, 2021-09-23:
Hey @pigeonhands , I noticed an issue in the way you are calling the subscribe function. You are providing the same subscription parameter for all connections. This has a problem because the bt_gatt_subscribe_params has the internal state variable `sys_snode_t node` which is used to put this in a linked list.
You are likely corrupting this list by providing the same `sub_params` for all connections.

Can you please try the updated code below, which has a unique `sub_param` per connection.

```c
static bt_gatt_subscribe_params sub_params[CONFIG_BT_CONN_MAX] = { 0 };

static uint8_t on_notify_data(struct bt_conn *conn,
                               struct bt_gatt_subscribe_params *params,
                                const void *data, uint16_t length){
    return BT_GATT_ITER_STOP;
}

void start_subscription(bt_conn * conn){
  sub_params.notify = on_notify_data;
  sub_params.value_handle = ctx.handles.measurement_value;
  sub_params.ccc_handle = ctx.handles.measurement_cccd;
  sub_params.value = BT_GATT_CCC_NOTIFY;
  atomic_set_bit(sub_params.flags, BT_GATT_SUBSCRIBE_FLAG_VOLATILE);
  int err = bt_gatt_subscribe(ctx.conn, &sub_params[bt_conn_index(ctx.conn)]);
  if (err == -ENOMEM) {
      //bug has happened
  }
}
```

If you still encounter the issue then could you please explain your reproduction steps in more details? There are still some things that are unclear to me.
Possibly a complete source file, and the exact steps you do with for example a phone?

pigeonhands, 2021-09-23:
@joerchan I only ever have one connection active at a time. The issue is that the [subscription variable](https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L73) in gatt.c is not being cleared properly. 

You can see in `gatt_sub_remove`, wich is called when the client disconnects, it removes the subscription from the  `sub_params` then if the list is empty, it removes it from the gatt.c `subscription`s:
https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L2706-L2716

However, `gatt_sub_remove` called on disconnect by itererating through the subscription list. If the list is empty, it will not be called.
https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L2880-L2891

The problem is `bt_gatt_unsubscribe` removes the subscription from the params list, but it does not remove it from the gatt.c `subcription` state. 
https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L4578-L4585
So when the disconnect happens, gatt_sub_remove will not be called for that client because its list is empty, but will still be in the gatt.c `subscription`

The `ENOMEM` comes from [gatt_sub_find](https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/subsys/bluetooth/host/gatt.c#L2720) returning null because all the items in the `subscriptions` are populated.

joerchan, 2021-09-24:
@pigeonhands Thanks for the update. Yes I can see that the issue happens now with a single connection connected at a time:

Reproduced this with 2 instances of tests/bluetooth/shell:
```
# shell 1:
bt init:
hrs simulate on

#shell 2:
bt init
```

Then loop this:
```
# shell 1:
bt oob

# shell 2:
bt connect <oob-addr>
gatt subscribe 17 15 
gatt unsubscribe
bt disconnect
```
Fails on the third loop iteration.

PR #38963 (2021-10-01): Bluetooth: Host: Fix resource leak in bt_gatt_unsubscribe
There are two simmilar functions for unsubscribing from GATT handles.

 - `gatt_sub_remove`, which is called on disconnect for every subscription
will free the `subscriptions` entry when the entry represents no
subscriptions.

 - `bt_gatt_unsubscribe`, called by the application, which forgets to
 free the `subscriptions` entry.

If all subscriptions grouped in a `subscriptions` entry are removed
using `bt_gatt_unsubscribe` before disconnect, there are no
subscriptions left to call `gatt_sub_remove` on. The `subscriptions`
entry is then  [... cut]
commit a5cf50e6 (2021-10-01): Bluetooth: Host: Fix resource leak in bt_gatt_unsubscribe

There are two simmilar functions for unsubscribing from GATT handles.

 - `gatt_sub_remove`, which is called on disconnect for every
 subscription will free the `subscriptions` entry when the entry
 represents no subscriptions.

 - `bt_gatt_unsubscribe`, called by the application, which forgets to
 free the `subscriptions` entry.

If all subscriptions grouped in a `subscriptions` entry are removed
using `bt_gatt_unsubscribe` before disconnect, there are no
subscriptions left to call `gatt_sub_remove` on. The `subscriptions`
entry is th [... cut]
commit 5969c3b9 (2021-10-02): Bluetooth: Host: Fix resource leak in bt_gatt_unsubscribe

There are two simmilar functions for unsubscribing from GATT handles.

 - `gatt_sub_remove`, which is called on disconnect for every
 subscription will free the `subscriptions` entry when the entry
 represents no subscriptions.

 - `bt_gatt_unsubscribe`, called by the application, which forgets to
 free the `subscriptions` entry.

If all subscriptions grouped in a `subscriptions` entry are removed
using `bt_gatt_unsubscribe` before disconnect, there are no
subscriptions left to call `gatt_sub_remove` on. The `subscriptions`
entry is th [... cut]
closed 2021-10-01 by cfriedt