zephyrproject-rtos/zephyr #38690: Wrong initialisation priority on different display drivers (eg. ST7735r) cause exception when using lvgl.

tfir1981, 2021-09-21:
**Describe the bug**
I am using the sample located to `sample/subsys/display/lvgl/` as reference for my code. 
I want to try out the lvgl framework on my hardware which consist of a display with the ST7735r-Controller. 
The following lines are my Kconfig:

```
#enable Display
CONFIG_DISPLAY=y
CONFIG_ST7735R=y

CONFIG_DISPLAY_LOG_LEVEL_ERR=y

CONFIG_LVGL_DISPLAY_DEV_NAME="DisplaySt7735r"

CONFIG_LVGL_HOR_RES_MAX=320
CONFIG_LVGL_VER_RES_MAX=240
CONFIG_LVGL_DPI=100

CONFIG_HEAP_MEM_POOL_SIZE=16384

CONFIG_LVGL=y
CONFIG_LVGL_USE_LABEL=y
CONFIG_LVGL_USE_CONT=y
CONFIG_LVGL_USE_BTN=y
CONFIG_LVGL_USE_THEME_MATERIAL=y
```

I compiled the sample code and load it into my MCU. While booting i got a Bus error exceptin. See logs below.

**Expected behavior**
I think the cause of this issue is lying [here](https://github.com/zephyrproject-rtos/zephyr/blob/357ec64b36cd84ca87660407e67eb0cb1cfe932b/drivers/display/display_st7735r.c#L578)

LVGL an the display driver have the same initialisation priority. When lvgl is initializing it cannot get the display device, because it is not initialized yet. If i set the init-level of the display driver to POST_KERNEL, the issue is fixed.

Is this the right way to fix the issue or are there any other switches to solve the problem.

**Impact**
It is not possible to use LVGL with the following display drivers:

- st7735r
- display_sdl
- st7789v

**Logs and console output**
```
[00:00:04.970,000] <err> os: ***** BUS FAULT *****
[00:00:04.970,000] <err> os:   Precise data bus error
[00:00:04.970,000] <err> os:   BFAR Address: 0x28f9c4f7
[00:00:04.970,000] <err> os: r0/a1:  0x28f9c4f7  r1/a2:  0x200118e0  r2/a3:  0x00000801
[00:00:04.970,000] <err> os: r3/a4:  0x28f9c4f7 r12/ip:  0x20011808 r14/lr:  0x0802a061
[00:00:04.970,000] <err> os:  xpsr:  0x21010000
[00:00:04.970,000] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
[00:00:04.970,000] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
[00:00:04.970,000] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
[00:00:04.971,000] <err> os: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
[00:00:04.971,000] <err> os: fpscr:  0x0803b230
[00:00:04.971,000] <err> os: Faulting instruction address (r15/pc): 0x0802a04e
[00:00:04.971,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:04.971,000] <err> os: Current thread: 0x20001a90 (SystemThread)
[00:00:05.105,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain PlatformIO
 - v2.6.0 until now

dleach02, 2021-09-21:
@tfir1981, please submit a PR to move this to POST_KERNEL per you suggestion. It isn't clear to me why this particular driver is being initialized in APPLICATION.

Only the display_st7735r.c and display_st7789v.c use APPLICATION level so I recommend having your PR change both of these to POST_KERNEL.

closed 2021-09-21 by nashif

tfir1981, 2021-09-22:
Good morning,
thank you all for taking care. I tested it with POST_KERNEL and it worked for me.

Kind regards
Thomas Fischer

PR #38722 (2021-09-21): display: st77xx: Move initialization to POST_KERNEL
These are the only two display drivers initializing in APPLICATION.
Dependency exposed with LVGL with initialization at same level and
priority.

Fixes #38690
commit 3ea7fbc6 (2021-09-21): display: st77xx: Move initialization to POST_KERNEL

These are the only two display drivers initializing in APPLICATION.
Dependency exposed with LVGL with initialization at same level and
priority.

Fixes #38690