zephyrproject-rtos/zephyr #38692: samples/tfm_integration: Some samples can't be built ("unexpected keyword argument 'rom_fixed'")

erwango, 2021-09-21:
**Describe the bug**
Compilation fails for some TFM samples on various targets:
Samples:
- samples/tfm_integration/psa_crypto
- samples/tfm_integration/tfm_ipc

Targets:
- mps2_an521_ns
- lpcxpresso55s69_ns
- stm32l562e_dk_ns

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
Compilation should succeed

**Impact**
Blocking

**Logs and console output**
```
Traceback (most recent call last):
  File "/local/mcu/zephyrproject/modules/tee/tfm/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py", line 128, in <module>
    wrap()
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/local/home/frq07517/.local/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/local/mcu/zephyrproject/modules/tee/tfm/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py", line 111, in wrap
    security_counter=security_counter)
TypeError: __init__() got an unexpected keyword argument 'rom_fixed'
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Linux
 - Toolchain (e.g Zephyr SDK, ...): zephyr-sdk-0.13.0
 - Commit SHA or Version used: v2.7.0-rc2-12-g9fa1f6edca

**Additional context**
Add any other context about the problem here.

microbuilder, 2021-09-21:
I've just tested this with 0.13.0 (though the OS X builds as a quick test since I have that setup), and with `west` it seems ro work fine:

```bash
$ west build -p auto -b mps2_an521_ns zephyr/samples/tfm_integration/psa_crypto/ -t run
...
*** Booting Zephyr OS build v2.7.0-rc2-41-g357ec64b36cd  ***
[00:00:00.001,000] <inf> app: app_cfg: Creating new config file with UID 0x1055CFDA7A
[00:00:03.779,000] <inf> app: att: System IAT size is: 545 bytes.
[00:00:03.779,000] <inf> app: att: Requesting IAT with 64 byte challenge.
[00:00:05.693,000] <inf> app: att: IAT data received: 545 bytes.
```

Switching to manually calling cmake with `ninja` I get:

```bash
$ cd samples/tfm_integration/psa_crypto
$ mkdir build && cd build
$ cmake -GNinja -DBOARD=mps2_an521_ns ..
$ ninja
$ ninja run
...
*** Booting Zephyr OS build v2.7.0-rc2-41-g357ec64b36cd  ***
[00:00:00.001,000] <inf> app: app_cfg: Creating new config file with UID 0x1055CFDA7A
[00:00:03.779,000] <inf> app: att: System IAT size is: 545 bytes.
[00:00:03.779,000] <inf> app: att: Requesting IAT with 64 byte challenge.
[00:00:05.693,000] <inf> app: att: IAT data received: 545 bytes.
```

With `make` I get:

```bash
$ cmake -DBOARD=mps2_an521_ns ..
$ make -j8
...
[ 18%] Built target linker_zephyr_prebuilt_script_target
make[1]: *** [zephyr/arch/arch/arm/core/aarch32/cortex_m/CMakeFiles/arch__arm__core__aarch32__cortex_m.dir/all] Error 2
[ 18%] Built target isr_tables
[ 18%] Built target arch__arm__core__aarch32__cortex_m__cmse
make[1]: *** [zephyr/arch/arch/arm/core/aarch32/CMakeFiles/arch__arm__core__aarch32.dir/all] Error 2
make: *** [all] Error 2
```

So this seems to be specific to `make` ... `ninja still works`.

Thanks for the heads up, though. I've only been testing with ninja and west.

microbuilder, 2021-09-21:
@erwango I'm not getting the same error with `make`, though. Your error seems to be related to the signing script, which may need to be updated to take into account recent changes to TF-M? Can you try running the `west` command and see if that works for you, and also making sure you're using the latest signing script?

It's the `imgtool` here: https://docs.zephyrproject.org/latest/guides/tfm/requirements.html#software-requirements

Something may have also changed there that isn't being handled properly (in the Zephyr repo or locally).

erwango, 2021-09-21:
@microbuilder Thanks for the quick feedback.

I didn't mention, but I was using `west build` actually.
On your advice, I've updated my pip3 packages (but I forgot to check what was the previous status).
And ... I'm getting the same error.

```
$ python3 -m pip show imgtool
Name: imgtool
Version: 1.6.0
Summary: MCUboot's image signing and key management
Home-page: http://github.com/JuulLabs-OSS/mcuboot
Author: The MCUboot committers
Author-email: dev-mcuboot@lists.runtime.co
License: Apache Software License
Location: /local/home/frq07517/.local/lib/python3.6/site-packages
Requires: cbor, click, intelhex, cryptography
Required-by: 
```

Note that I'm getting a strange warning when calling pip3 directly. Not sure if this can have an impact.
```
$ pip3 show imgtool
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
```

erwango, 2021-09-21:
> u try updating to the most recent version of imgtool? I think we might need to update the documentation as well. I see you're on 1.6.0, but this may be too old for 1.3.0. I'll look at the TF-M docs myself to see what they require.

The TFM doc you pointed actually specifies "imgtool>=1.6.0"; so it might require an update indeed.
Anyway, updated to v1.7.0 now, and the error persists.

SebastianBoe, 2021-09-21:
Strange that in the issue description "make" is invoked, but this shows up the log output:

> ninja: build stopped: subcommand failed.

SebastianBoe, 2021-09-21:
Could not reproduce ninja vs Make problem. 

r```
m -rf build/ && cmake -H. -Bbuild -DBOARD=mps2_an521_ns -G"Unix Makefiles"
Including boilerplate (Zephyr base): /home/sebo/ncs/zephyr/cmake/app/boilerplate.cmake
CMake Deprecation Warning at /home/sebo/ncs/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
  The OLD behavior for policy CMP0079 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  /home/sebo/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/sebo/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Application: /home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc
-- Zephyr version: 2.7.0-rc2 (/home/sebo/ncs/zephyr), build: v2.7.0-rc2-41-g357ec64b36cd
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: mps2_an521_ns
-- Cache files will be written to: /home/sebo/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.13.0 (/home/sebo/zephyr-sdk-0.13.0)
-- Found dtc: /home/sebo/zephyr-sdk-0.13.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/sebo/ncs/zephyr/boards/arm/mps2_an521/mps2_an521_ns.dts
-- Generated zephyr.dts: /home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build/zephyr/dts.cmake
Parsing /home/sebo/ncs/zephyr/Kconfig
Loaded configuration '/home/sebo/ncs/zephyr/boards/arm/mps2_an521/mps2_an521_ns_defconfig'
Merged configuration '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/prj.conf'
Configuration saved to '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build/zephyr/.config'
Kconfig header saved to '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sebo/zephyr-sdk-0.13.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at /home/sebo/ncs/zephyr/CMakeLists.txt:722 (message):
  No SOURCES given to Zephyr library: drivers__pinmux

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build
sebo@sebo:~/ncs/zephyr/samples/tfm_integration/tfm_ipc$ make -C build/
make[2]: Leaving directory '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build'
make[2]: Entering directory '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build'
[ 97%] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[ 98%] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[100%] Building C object zephyr/CMakeFiles/zephyr_final.dir/dev_handles.c.obj
[100%] Linking C executable zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16932 B       512 KB      3.23%
            SRAM:        7224 B         1 MB      0.69%
           SRAM1:          0 GB         4 MB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
Generating files from zephyr.elf for board: mps2_an521_ns
make[2]: Leaving directory '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build'
[100%] Built target zephyr_final
make[1]: Leaving directory '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build'
make: Leaving directory '/home/sebo/ncs/zephyr/samples/tfm_integration/tfm_ipc/build'
sebo@sebo:~/ncs/zephyr/samples/tfm_integration/tfm_ipc$ git show HEAD
commit 357ec64b36cd84ca87660407e67eb0cb1cfe932b (HEAD, us/main)

```

microbuilder, 2021-09-21:
@erwango Just to be sure, have you tried a `west update` to make sure you're using the right version of the TF-M module, and that `bl2/ext/mcuboot/scripts/wrapper/wrapper.py` is in sync with the rest of Zephyr?

erwango, 2021-09-21:
> @erwango Just to be sure, have you tried a `west update` to make sure you're using the right version of the TF-M module, and that `bl2/ext/mcuboot/scripts/wrapper/wrapper.py` is in sync with the rest of Zephyr?

I think it's ok from this point of view:
```
=== updating mcuboot (bootloader/mcuboot):
HEAD is now at 70bfbd21 merge: synchronize up to 85da97f + cherry-pick upstream bugfixes for LTS 2.7.0
```

erwango, 2021-09-21:
Compilation succeeds after imgtool uninstall/reinstall.
Closing the issue. Thanks @microbuilder for the support.

closed 2021-09-21 by erwango