zephyrproject-rtos/zephyr #38694: Disabling NET_CONFIG_AUTO_INIT does not require calling net_config_init() manually in application as mentioned in Zephyr Network Configuration Library documentation

vntsn, 2021-09-21:
**1.** I have disabled NET_CONFIG_AUTO_INIT in prj.conf file as I want to do network configuration and initialization in application. As per documentation "If you want to use the network configuration library but without automatic initialization, you can call net_config_init() manually". Reference: https://docs.zephyrproject.org/2.6.0/reference/networking/net_config.html
**But networking is working fine without calling net_config_init() manually, So the documentation contradicts with code behavior.**

In application I am using both IP allocation methods alternatively, i.e DHCP and static, and running TCP echo server.
For DHCP initialization below function are called:
iface = net_if_get_default();
net_dhcpv4_start( iface );

For Static IP initialization below function is called:
net_if_ipv4_addr_add( net_if_get_default(), &ipv4_addr, NET_ADDR_MANUAL, 0U );

Not calling net_config_init() or any other initialization function in application but still networking is getting initialized when application starts, despite of this I tried calling net_config_init() in application but its definition is not found if NET_CONFIG_SETTINGS is disabled.
 **Please help me to understand how the networking is getting initialized when NET_CONFIG_AUTO_INIT is disabled, net_config_init() not called manually and what configuration parameters can be controlled from application in this case. I am not calling net_config_init(), will it impact anything in networking?
What is the right way to configure networking in application? **. 

Network configuration in prj.conf file:
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_NET_CONFIG_SETTINGS=n
CONFIG_NET_CONFIG_AUTO_INIT=n

**2.** **I am getting below error messages every time while running the application and also when ethernet cable is unplugged and plugged back. It is not impacting the functionality but what is the cause of these errors and how to fix it?**
[image]

**3.** **How much time is expected to get the interface completely up when ethernet cable is connected while application is running** instead of connecting cable before starting application. I have observed that it takes few seconds to get the interface up when Ethernet cable is connected after starting the application.

**Environment:**
 - OS: Windows
 - IDE : Visual Studio Code-PlatformIO
 - Zephyr version: 2.6.0

jukkar, 2021-09-22:
> "If you want to use the network configuration library but without automatic initialization, you can call net_config_init() manually"

Originally the idea was that if one wants to enable the settings library (`CONFIG_NET_CONFIG_SETTINGS=y`), then the auto init can be disabled, but in which case the `net_config_init()` **can** be called. Note that calling that function is optional here.
If the settings library is not enabled, then of course the function should not be called in the first place.

rlubos, 2021-09-22:
I think you're confusing the purpose of the `NET_CONFIG` library. It does not affect the network initialization per se - even if the library is disabled, the networking subsystem should be initialized and ready to go. The only difference is, that it might not be configured to operate in the network out-of-the-box. 

The `NET_CONFIG` library purpose, in simplification, is to configure IP addresses on the interfaces. But you can also do this manually, w/o engaging `NET_CONFIG` library (from what you mention, you run DHCP manually on your interface). Or you can run the `NET_CONFIG` initialization manuallly (the aforementioned CONFIG_NET_CONFIG_AUTO_INIT=n), but in this case the only difference is that you need to call `net_config_init()` by yourself, normally the system does that for you during bootup. And as Jukka says, you still need `NET_CONFIG` library enabled in order to do that.

Note, that the documentation does not state that you **have to** call `net_config_init()` in order to use the networking, it just states you have this option to run the configuration library manually - still, it's not mandatory. Therefore, I don't think there's a bug anywhere in this behaviour, I'd rather set the issue as a question.

> 2. I am getting below error messages every time while running the application and also when ethernet cable is unplugged and plugged back. It is not impacting the functionality but what is the cause of these errors and how to fix it?

But is this permanent? The network interface is likely put down when you plug off the cable, which is also indicated by the errno returned by `net_recv_data()` - `ENETDOWN`. Once the cable is reattached, and the network interface is up again it should be possible to receive data again (which I assume takes place given the DHCP log below the errors?).

> How much time is expected to get the interface completely up when ethernet cable is connected while application is running instead of connecting cable before starting application. I have observed [...]

This one I don't know, I guess it depends on particular Ethernet driver.

vntsn, 2021-09-23:
Thanks @rlubos and @jukkar! Based on your comment I believe that the way I have configured/initialized DHCP is correct and no need to call net_config_init().

> But is this permanent? The network interface is likely put down when you plug off the cable, which is also indicated by the errno returned by net_recv_data() - ENETDOWN. Once the cable is [...]

Yes, data send/receive take place when the interface is up again. These errors are observed every time on application startup and when ethernet cable is unplugged/plugged. These errors are observed even when auto config is used (instead of manual configuration) i.e NET_CONFIG_AUTO_INIT=y.

jukkar, 2021-09-23:
I am not sure what you are trying to accomplish but after you mentioned plugging/unplugging cable etc. and some problems, there is an network management API that can be used to monitor network interface up/down, IP connectivity and other relevant events in the system. So if you for example want to do something when network interface comes up, there is an easy way to act on these events and do something application specific things.

vntsn, 2021-09-27:
Yes, I am using the network management API for below events, 
NET_EVENT_IF_UP
NET_EVENT_IF_DOWN
NET_EVENT_IPV4_ADDR_ADD
Though I was curious to know specifically about these unexpected RX queue errors.

Thanks @jukkar @rlubos

closed 2021-09-27 by vntsn