zephyrproject-rtos/zephyr #38697: The coap_server sample is missing the actual send in the retransmit routine

AllemanMa, 2021-09-21:
**Describe the bug**
If a device can't reach it's destination, it will get into `retransmit_request` for several times to decrement its retries, however no actual send is performed which makes the retransmit request useless.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the coap_server sample
2. Observe the resource i.e. `coap get -o coap://xx.xx.xx.xx/obs`
3. Stop the coap client
4. Wait until the coap server gets into the retransmit_request

**Expected behavior**
I would expect the coap server to actual resend a message instead of just decrementing the retries.

**Impact**
It leaves the coap server retry mechanism useless.

**Environment (please complete the following information):**
 - Zephyr v2.6.0

cfriedt, 2021-09-21:
@AllemanMa - are you able to submit any PRs for this or other coap issues? tests welcome as well.

rlubos, 2021-09-22:
I've submitted a fix in https://github.com/zephyrproject-rtos/zephyr/pull/38748, indeed the retransmission mechanism was broken.

PR #38748 (2021-09-27): samples: net: coap_server: Fixes for retransmission and observer
Several fixes for the sample for better user experience:
 * Fix retransmission mechanism, to actually send the retransmitted message and fix the retransmission timeout calculation.
 * Remove observer on notification timeout to release resources.
 * Finetune some of the sample parameters, to improve sample responsiveness.

Fixes #38697
Fixes #38700
Fixes #38701
Fixes #38702
closed 2021-09-27 by nashif