zephyrproject-rtos/zephyr #38700: Observer of coap server not working properly

AllemanMa, 2021-09-21:
**Describe the bug**
If I try to observe the `/obs` resource from my pc, I do get the desired behaviour of the counter being incremented and transmitted every 5 seconds. If I however open a second terminal and observe the `/obs` resource for a second time as well, the first observer keeps receiving counter updates however my second observer doesn't. It received the counter value only a single time. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build the coap_server sample
2. Observe the resource i.e. coap get -o coap://xx.xx.xx.xx/obs
3. Open another terminal and observe the resource again coap get -o coap://xx.xx.xx.xx/obs
4. Only first observer keeps receiving counter updates
5. The second observer will only receive a single counter update

**Expected behavior**
Both observers must receive the same counter updates.

**Impact**
I am not able to have 2 observers observing the same resource.

**Logs and console output**
Observer 0

```
coap get -o coap://xx.xx.xx.xx/obs
(2.05)  Counter: 0

(2.05)  Counter: 1

(2.05)  Counter: 2

```

Observer 1

```
coap get -o coap://xx.xx.xx.xx/obs
(2.05)  Counter: 0
```

Coap server

```
*** Booting Zephyr OS build zephyr-v2.6.0  ***

[00:00:00.057,000] <inf> net_config: Initializing network
[00:00:00.057,000] <inf> net_config: Waiting interface 1 (0x80000dc8) to be up...
[00:00:03.011,000] <inf> net_config: Interface 1 (0x80000dc8) coming up
[00:00:03.011,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:03.011,000] <inf> net_config: Running dhcpv4 client...
[00:00:12.066,000] <inf> net_dhcpv4: Received: xx.xx.xx.xx
[00:00:12.066,000] <inf> net_config: IPv4 address: xx.xx.xx.xx
[00:00:12.066,000] <inf> net_config: Lease time: 604800 seconds
[00:00:12.066,000] <inf> net_config: Subnet: xx.xx.xx.xx
[00:00:12.066,000] <inf> net_config: Router: xx.xx.xx.xx
[00:00:12.066,000] <dbg> net_coap_server_sample.main: Start CoAP-server sample
[00:00:14.349,000] <inf> net_coap_server_sample: *******
[00:00:14.350,000] <inf> net_coap_server_sample: type: 0 code 1 id 18191
[00:00:14.350,000] <inf> net_coap_server_sample: *******
[00:00:14.350,000] <dbg> net_coap_server_sample: Response
                                                 64 45 47 0f 98 f9 a6 54  61 02 60 ff 43 6f 75 6e |dEG....T a.`.Coun
                                                 74 65 72 3a 20 30 0a                             |ter: 0.
[00:00:16.694,000] <inf> net_coap_server_sample: *******
[00:00:16.694,000] <inf> net_coap_server_sample: type: 0 code 1 id 9353
[00:00:16.694,000] <inf> net_coap_server_sample: *******
[00:00:16.694,000] <dbg> net_coap_server_sample: Response
                                                 64 45 24 89 6c 20 c4 f0  61 02 60 ff 43 6f 75 6e |dE$.l .. a.`.Coun
                                                 74 65 72 3a 20 30 0a                             |ter: 0.
[00:00:21.694,000] <dbg> net_coap_server_sample: Response
                                                 44 45 43 8a 98 f9 a6 54  61 03 60 ff 43 6f 75 6e |DEC....T a.`.Coun
                                                 74 65 72 3a 20 31 0a                             |ter: 1.
[00:00:21.748,000] <dbg> net_coap_server_sample: Response
                                                 44 45 43 8b 6c 20 c4 f0  61 03 60 ff 43 6f 75 6e |DEC.l .. a.`.Coun
                                                 74 65 72 3a 20 31 0a                             |ter: 1.
[00:00:24.070,000] <inf> net_coap_server_sample: RETRANSMIT (17291)(4)
[00:00:26.748,000] <dbg> net_coap_server_sample: Response
                                                 44 45 43 8c 98 f9 a6 54  61 04 60 ff 43 6f 75 6e |DEC....T a.`.Coun
                                                 74 65 72 3a 20 32 0a                             |ter: 2.
[00:00:26.802,000] <dbg> net_coap_server_sample: Response
                                                 44 45 43 8d 6c 20 c4 f0  61 04 60 ff 43 6f 75 6e |DEC.l .. a.`.Coun
                                                 74 65 72 3a 20 32 0a                             |ter: 2.

```

**Environment (please complete the following information):**
 - Observers: Windows
 - Zephyr V2.6.0

cfriedt, 2021-09-21:
@rlubos - are you able to triage this further? Is this expected behaviour as per Zephyr's docs?

rlubos, 2021-09-22:
This one actually I cannot reproduce, I can observe a resource with 2 observers w/o any issue. This might be related however to broken retransmission mechanism reported in other issue. The sample may easily run out of resources, preventing further observations (but it's only observable to me if I start/stop other observations. I'll try to propose a combo PR to address all those issues in the sample.

PR #38748 (2021-09-27): samples: net: coap_server: Fixes for retransmission and observer
Several fixes for the sample for better user experience:
 * Fix retransmission mechanism, to actually send the retransmitted message and fix the retransmission timeout calculation.
 * Remove observer on notification timeout to release resources.
 * Finetune some of the sample parameters, to improve sample responsiveness.

Fixes #38697
Fixes #38700
Fixes #38701
Fixes #38702
closed 2021-09-27 by nashif