zephyrproject-rtos/zephyr #38715: DT Macro for counting nodes

aaronemassey, 2021-09-21:
**Is your feature request related to a problem? Please describe.**

I have a use-case where I need to know the length of children nodes or nodes with a compatible string.

**Describe the solution you'd like**

Two DT macros like:

* `DT_COUNT_STATUS_OKAY(compat)`
* `DT_COUNT_CHILDREN(NODE_ID)`

**Describe alternatives you've considered**
Currently my solution is to write macro functions like:

```
#define _COUNT_STATUS_OKAY_COUNTER() 1 +
#define COUNT_STATUS_OKAY(compat) (DT_FOREACH_STATUS_OKAY(compat, _COUNT_STATUS_OKAY_COUNTER) 0)
```

But this seems kind of ugly and instead something that maybe should be usable from the DT macros.

**Additional context**
I don't think any additional context is needed besides the example I supplied.

Thank you for all the work that has been put into Zephyr.

aaronemassey, 2022-05-23:
I'll have a tiny pull-request for this fairly soon.

TomzBench, 2023-03-11:
I currently embed a struct of dt_spec nodes which represent ethernet ports for a switch product that can have N number of ports. So  this  feature allows me to embed these nodes properly.

mbolivar-nordic, 2023-03-17:
@aaronemassey thanks for volunteering to add this. Please use `DT_NUM_` instead of `DT_COUNT_` for consistency:

```
$ rg '#define DT_NUM' include/zephyr/devicetree.h
1620:#define DT_NUM_RANGES(node_id) DT_CAT(node_id, _RANGES_NUM)
2146:#define DT_NUM_REGS(node_id) DT_CAT(node_id, _REG_NUM)
2239:#define DT_NUM_IRQS(node_id) DT_CAT(node_id, _IRQ_NUM)
2955:#define DT_NUM_INST_STATUS_OKAY(compat)			\
```

mbolivar-nordic, 2023-03-17:
I think you only need DT_NUM_CHILDREN, and DT_NUM_INST_STATUS_OKAY already exists for your other example.

aaronemassey, 2023-03-29:
> I'll have a tiny pull-request for this fairly soon.

Sorry for letting this drop. It only recently came back on my radar due to the issue activity :) Looking to have a PR up this or next week.

aaronemassey, 2024-04-24:
> Hi @aaronemassey Is there a PR? I'm in need of this too.

Bah, I should never commit to things so quickly like in https://github.com/zephyrproject-rtos/zephyr/issues/38715#issuecomment-1489242523. Things have been busy, but we're all busy :)

Will *try* to have something this week/end. @swift-tk, you're welcome to add it if you want it sooner. It should basically be @mbolivar-nordic suggestions in https://github.com/zephyrproject-rtos/zephyr/issues/38715#issuecomment-1474433804.

swift-tk, 2024-04-25:
Hi @aaronemassey 
I think I have a better solution see https://github.com/zephyrproject-rtos/zephyr/pull/71933.

closed 2024-11-19 by rruuaanng