zephyrproject-rtos/zephyr #38719: mcuboot configured to use scratch partition (with QSPI nRF52840) does not work

lairdjm, 2021-09-21:
**Describe the bug**
Some devices such as the nRF52840 require buffers to be 4-byte aligned and also lengths to be 4-byte aligned, mcuboot when built to use the scratch partition on an nRF52840 whereby one slot resides in QSPI does not work. This seems to be because mcuboot will, in that configuration, make multiple 1-byte reads which are not supported, the data it gets back is either empty or garbage, and as a result it hits an assertion and panics.

**To Reproduce**
Any mcuboot-configured application which requires 4-byte length aligned reads and is configured with mcuboot in a scratch mode, the initial application will load, perform an update with the second slot with mcuboot and the application will not run.

**Expected behavior**
For the update to work

**Impact**
Showstopper. The reason being that in swap without scratch mode, the upgrade takes about 4 times longer in a production release mode that it does using scratch with a small hacked, fully debug output version that I've quickly made to track the issue down and work around it

**Logs and console output**
Don't have the log any more but the output with logging enabled is roughly something like ./boot/bootutil/src/swap_scratch.c on line 600 assertion error, boot_copy_region() rc is not 0

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.1
 - Commit SHA or Version used: ncs 1.6.1 using zephyr 2.6 but this issue has persisted for multiple zephyr versions now

**Additional context**
By "hacking around" the way I'm working around the issue if checking if flash_area_read is reading less than 4 and if so reading 4 to a temp. buffer and then copying the amount of data requested to the actual buffer

cfriedt, 2021-09-21:
Is this an MCU Boot bug or Zephyr bug?

lairdjm, 2021-09-21:
@cfriedt the flash map files are in zephyr itself, used by both zephyr and mcuboot, so based on that would say a zephyr issue. Although maybe zephyr has code that deals with the alignment which isn't built into the mcuboot code

lairdjm, 2021-09-22:
Seems like my mini fix didn't fix it. I've got a known app_update.bin file which I'm transferring via an app and letting it upgrade in swap with scratch mode, after this I get the "no image in primary slot" error, I've taken an image of the flash after this and compared it with the app_update.bin file, mcuboot is built for 0x0 - 0x10000 and the app from 0x10000 - 0xE8000, the scratch area is in internal flash from 0xF8000 - 0xFC000, I've put both files into HxD and see differences at numerous offsets e.g. 0x634C, 0x6350, 0xA340, 0xA344, 0xE334, 0xE338, 0x12328, 0x1232C - these all seem to be in pairs and positions are 0x3FF4 different in offsets from each other, so that seems linked to the scratch partition being 0x4000 in size. I've tried aligning buffers and increasing the mcuboot buffer size to 4KB (to match the nRF52840 sector size) and added some debug messages, log on https://pastebin.com/RpyPF1SG

de-nordic, 2021-10-04:
I have been trying to reproduce this on nrf52840dk_nrf52840 with slot-1 set on mx25, but it works for me.
I have included overlay that defines outer slot for me and .config taken from the mcuboot build and the smp_svr I am using for tests. 
[mcuboot_scratch.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/7279891/mcuboot_scratch.config.txt)
[outer-slot.overlay.txt](https://github.com/zephyrproject-rtos/zephyr/files/7279892/outer-slot.overlay.txt)
[smp_svr.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/7279893/smp_svr.config.txt)

lairdjm, 2021-10-04:
That's really odd, I've tried the same smp_svr sample app with default mcuboot configuration (with the change to use the scratch) and as you see when you try it, it does work, but I've then tried again with our application and it's not working, it starts up, I transfer an image via the UART and then test that image, reboot the module and it swaps the image over for about a minute before saying it can't find a bootable image because the primary slot is not valid, I then reboot and it reverts and I get the same message about being unable to find a bootable image because the primary slot is not valid, I then reboot one more time, it does a large erase and displays the same message.

de-nordic, 2021-10-05:
This does not seem to be problem in Zephyr, can we convert this to discussion and close the issue for now?

lairdjm, 2021-10-05:
After dumping nRF connect SDK entirely and going with mainline zephyr I can confirm it does seem that the issue here is not one issue but 2. One in mcuboot which I've opened here https://github.com/mcu-tools/mcuboot/issues/1165 and would like to get resolved for zephyr 2.7 because we have multiple boards we want to support with this going forward, and the other issue seems to be with the nRF connect SDK for some reason which is something that needs looking into and reporting on the devzone at another time.

closed 2022-01-24 by github-actions[bot]