zephyrproject-rtos/zephyr #38745: Bluetooth when configured for extended advertising does not limit advertisement packet size if a non-extended avertisement is used

lairdjm, 2021-09-22:
**Describe the bug**
Using nRF connect SDK 1.6.1 based on zephyr 2.6 with an nRF52840, in subsys/bluetooth/host/adv.c it seems that when advertisements and scan reports are generated when the code has been build with extended advertising support but starting using NON-extended advertising, the maximum packet sizes are incorrectly set to the extended advertising lengths, so if you have a device name of length 50 characters, it will put all 50 characters into the packet and then complain that the packet is too large even though the shortened name should be used instead. In the `hci_set_ad_ext` function is:
```
        err = set_data_add(set_data->data, BT_HCI_LE_EXT_ADV_FRAG_MAX_LEN,
                           ad, ad_len, &set_data->len);
```
But the second parameter is valid only for extended advertisements, for non-extended advertisements the hard limit is 31.

**To Reproduce**
Create an application with support for extended advertising and set a large (e.g. 50 character) device name and advertise in legacy mode

**Expected behavior**
For the device name to be limited and maximum packet size to be 31 bytes

**Impact**
Showstopper, legacy advertising is broken

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: ncs 1.6.1 using zephyr 2.6

lairdjm, 2021-09-22:
Fixed locally for me in subsys/bluetooth/host/adv.c by changing (in hci_set_ad_ext):
```
	err = set_data_add(set_data->data, BT_HCI_LE_EXT_ADV_FRAG_MAX_LEN,
			   ad, ad_len, &set_data->len);
```
to
```
	err = set_data_add(set_data->data,
			   (((*adv->flags) & BT_LE_ADV_OPT_EXT_ADV) !=
			   BT_LE_ADV_OPT_EXT_ADV ?
				BT_GAP_ADV_MAX_ADV_DATA_LEN :
				BT_HCI_LE_EXT_ADV_FRAG_MAX_LEN),
			   ad, ad_len, &set_data->len);
```

If no one has an issues with the fix, I'll get a PR up in a few hours

lairdjm, 2021-09-22:
Seems that in 2.7 there is the idea of fragmented packets now which isn't in 2.6 so not sure how this would apply to the fragmented part, if at all.

carlescufi, 2021-09-23:
@lairdjm please send a PR and we'll review it there.

joerchan, 2021-09-23:
@lairdjm `atomic_test_bit(adv->flags, BT_ADV_EXT_ADV)` is the correct way to test if it is legacy or extended advertising set.

You shouldn't need to handle anything differently with fragmented supported, since the legacy advertising data will always fit inside one fragment.

closed 2021-09-27 by nashif