zephyrproject-rtos/zephyr #38747: data/json: encoding issues with array in object_array

uschlaepfer, 2021-09-22:
**Describe the bug**
if you try to encode a obj_array which contains an array only the first element is encoded in the string.

**To Reproduce**
I have created an example for this issue (files attached [json_issue.zip](https://github.com/zephyrproject-rtos/zephyr/files/7211008/json_issue.zip))

The printed result is:
`{"data":{"namePkg":[{"name":["John0","Adam0","Julia0","Laura0"]},{"name":[]}],"numberPkg":[{"number":["123","456"]},{"number":[]},{"number":[]}]}}`

**Expected behavior**
The expected printed result is:
`{"data":{"namePkg":[{"name":["John0","Adam0","Julia0","Laura0"]},{"name":["John1","Adam1","Julia1"]}],"numberPkg":[{"number":["123","456"]},{"number":["789","987"]},{"number":["654","286","217","984"]}]}}`

**Impact**
I have to send similar json strings -> showstopper

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK 2.6.0 + west
 - v2.6.0

carlescufi, 2021-09-22:
@mrfuchs can you please take a look?

d-soliveira, 2021-09-24:
Hi there,
Similar problem here.
**Problem**:
`{"Objects":[{"Obis":"FF","Attributes":[]},{"Obis":"�␇","Attributes":[]}]}`
**Expected**:
`{"Objects":[{"Obis":"FF","Attributes":[{"Number":99,"Data":"ff"}]},{"Obis":"FF","Attributes":[{"Number":99,"Data":"ff"}]}]}`

It seams, for some reason, on the second array element the variable `n_elm` overflows.

martinjaeger, 2021-09-26:
Ping also @Sir-Branch.

@nashif can you add @Sir-Branch to the collaborators in GitHub (see #38478) so that he can be assigned to JSON-related issues and do valid PR reviews?

Sir-Branch, 2021-09-27:
I know this might be a active blocker for some users. I was able to reproduce, and agree that this is an issue.

In my opinion this is a symptom of an underlying shortcoming in the architecture, it wasn't really thought for use of nested objects/array from the get go (see the [original authors blog post](https://tia.mat.br/posts/2017/03/01/parsing_json.html)). 

Support for nested objects seems more like a patch in my opinion. Handling nested objects and structures in a cleaner fashion is something I have on my radar, but haven't really come up with a way yet.

I'll try to see if I'm able  to figure out what's going on

kowalewskijan, 2021-09-27:
@Sir-Branch I pushed a proposal for this issue: https://github.com/zephyrproject-rtos/zephyr/pull/38888 please take a look.

closed 2022-04-20 by github-actions[bot]