zephyrproject-rtos/zephyr #38755: Integrate U8G2 into zephyr

DhruvaG2000, 2021-09-22:
## Origin

Name of project hosting the original open source code: 
[U8g2: Library for monochrome displays, version 2](https://github.com/olikraus/u8g2)

## Purpose

U8g2 is a monochrome graphics library for embedded devices. U8g2 supports monochrome OLEDs and LCDs

## Mode of integration

I'd like to integrate this external component as a module, as it will make using OLED and other display modules much easier to use without needing to add this as a seperate library into the zephyr project. 
The module repo name could still simply be U8G2

## Pull Request

Pull request : none currently

## Description

- Includes all graphics procedures (line/box/circle draw).
- Supports many fonts. (Almost) no restriction on the font height.
- Requires some memory in the microcontroller to render the display.

What is its primary functionality?

Supported a wide variety of Display Controllers and also provide easy interface with many fonts and symbols to choose from.

What problem are you trying to solve? 

This will make interfacing OLED modules a breeze using zephyr

Why is this the right component to solve it?

It is very widely used, and also is very fast, lightweight and also fewer bugs compared to all the other OLED and display drivers out there.
This will also make maintaining display and OLED drivers much easier than needing to individually write a driver for each and every other display and it's controller.

# Dependencies

What other components does this package depend on?

This is a self sufficient library.

## Revision

1ba43d1a4d460a35ad9ddded28f240ffc65ddee0 is the latest and has been tested by me personally on my nano ble sense 33. I can provide video proof if required.

## License

``new-bsd license (two-clause bsd license)``

jfischer-no, 2021-09-23:
> It is very widely used, and also is very fast, lightweight and also fewer bugs compared to all the other OLED and display drivers out there.
> This will also make maintaining display and OLED drivers [...]

Compared to what exactly?

Anyway, NACK, we have display controller driver API and controller driver in Zephyr. We do not need more dependencies. It is about time that the development of own graphic subsystem is encouraged.

DhruvaG2000, 2021-09-23:
> Anyway, NACK, we have display controller driver API and controller driver in Zephyr. We do not need more dependencies. It is about time that the development of own graphic subsystem is encouraged.

Understood, and looking forward to seeing a new _zephyr_ graphic subsystem

gmarull, 2021-09-23:
@DhruvaG2000 feel free to open a proposal with a port of the library to Zephyr. At a quick glance, it doesn't look to be as portable/well-designed as others like LVGL, so I'm not sure if a _library_ like this has a chance to suceed outside the Arduino ecosystem.

ck-telecom, 2021-09-27:
There seems a similar zephyr/subsys/fb with fonts(mono similar with dot matrix display), I think it is a good idea to port to this directory with more fonts and more draw APIs.

rivimey, 2022-01-16:
I had also decided that u8g2 was a good match for my use case, before I started looking at Zephyr at all. I am driving a SSD1320Z based 320x132 greyscale OLED. The Zephyr codebase doesn't support it at all - no driver, no support for greyscale framebuffers, and even the ssd1306 driver has multiple bugs reported.

Following the comments earlier in this issue, I looked into the LVGL library and found that it is serving a different use case. While u8g2 is mostly about driving (many, many) displays efficiently, and offers some limited graphics primitives on top of that, LVGL is mostly about offering a comprehensive GUI option on top of a framebuffer. The device driver options within it are very limited indeed (and don't include e.g. greyscale displays).

There is a small area of overlap: u8g2 does have some drawing functions, LVGL does have some device functions, but the two libraries are for the most part serving different needs. For Zephyr, the main downside of using u8g2 is that it doesn't do RGB colour. For LVGL its hardware support options are severly limited both in chip support and data formats.

I can agree that adding u8g2 to Zephyr isn't a brilliant match, but on the other hand with so much development happening on Zephyr elsewhere and so much knowledge & code embedded in u8g2 I see no prospect of a Zephyr native version of this area of code being available in even the medium term. Consequently I would suggest a compromise is found.

@DhruvaG2000 I would be interested to see how you did your integration, even if not an official version... is it available anywhere (pref as a PR of some sort)?

DhruvaG2000, 2022-01-16:
> @DhruvaG2000 I would be interested to see how you did your integration, even if not an official version... is it available anywhere (pref as a PR of some sort)?

I was planning on using zephyr but found more driver support in the Arduino platform itself, hence I did not integrate U8G2 with zephyr. However if you are still interested you can see my project [here](https://github.com/DhruvaG2000/SmartDuinoSensorHub)

romkell, 2022-05-31:
In addition to the gmarull's link above to have the arguments here:

Issues is see with the current setup in Zepyhr OS
- LVGL integrated in Zephyr OS is full featured and powerful but consumes a lot of flash for small MCU devices (>64kB)
- character frame buffer is very basic (no drawing, only multiple of 8 character size without special care on drawing)
- For small MCUs LVGL is too heavy weight and CFB is very basic
- The only OLED driver included today is the SSD1306, no other simple monochrome drivers available
- There is a lack of different monochrome LCD / OLED drivers in Zephyr OS today

Advantages of u8g2 as module into Zephyr OS since it
- it has a small footprint (some 9kB flash)
- supports a variety of monochrome LCD / OLED drivers
- supports simple drawing capabilities
- can be used as character only as u8x8 (less RAM requirements)
- as an external module we would get a lot of it in one step

Concerns and Unresolved Questions
- Since u8g2 implements the display driver itself, I assume the Zephyr OS display interface would be by-passed
- The i2c, spi or other interface would have to be ported in u8g2 somehow (a RT-Thread port exists in u8g2)
- Does not really (seem) to fit into Zephyr OS architecture this way (without having analyzed in detail)

Alternatives (to us - my company)
- Integrate Zephyr OS as port in u8g2 for particular versions of Zephyr OS
- Integrate in company internal framework

romkell, 2022-06-01:
> Anyway, NACK, we have display controller driver API and controller driver in Zephyr. We do not need more dependencies. It is about time that the development of own graphic subsystem is encouraged.

An alternative for Zephyr OS I see going into the direction of "develop own graphic subsystem" which would be the nice and beautiful way:
- take u8g2's drivers and refactor them underneath Zephyrs display interface
- take the glyph and drawing part, refactor them and add them to the CFB (extend CFB, or replace it)
- find sponsor for the relatively large effort
- find a maintainer to it (to all the ported code)

The pragmatic approach is using it as is and integrate as Zephyr OS module (with all the numerous compiler switches, possibly mapping them to KConfig). Maintenance then is reduced to the integration and possibly upgrading to a newer version every now and then, I feel.

Company internal we have not decide yet whether to add a SH1107 driver (which ever it will be in the end) in to Zephyr OS and to continue to use CFB, as limited as it is or to switch to u8g2.
As project driven as "my" company is and as an end user of Zephyr OS, I can justify the pragmatic way or an ultra light version of the architectural nice version (add SH1107). The full extend I will not get through.

gmarull, 2022-06-01:
I think that nothing prevents to create a new module that integrates Zephyr with u8g2, one can just create it and use it on their apps. Whether it needs to be maintained by upstream Zephyr or not is a separate question. I've voiced my opinion many times on this aspect: the community should be able to create and distribute Zephyr modules on their own, there's no need to go upstream to make a module work. However, I think it's reasonable to ask upstream a way to promote or make such modules visible, so that application developers are aware of them.

DhruvaG2000, 2022-06-01:
> However, I think it's reasonable to ask upstream a way to promote or make such modules visible, so that application developers are aware of them

Perhaps something like [an awesome list](https://github.com/golioth/awesome-zephyr-rtos) can be maintained upstream which lists all modules approved and tested by zephyr.

> find sponsor for the relatively large effort

Well this might very well be a good idea for next year's Google Summer of Code. 

Also, if there is good progress on an Arduino-Core module for Zephyr then perhaps we could directly use U8G2 lib without having to refactor it's drivers into zephyr.

jfischer-no, 2022-07-06:
>Alternatives (to us - the company I work for)
>
>    Integrate Zephyr OS as port in u8g2 for particular versions of Zephyr OS
>    Integrate in company internal framework

There is nothing that prevents you from doing this from the technical side. But you might want to take a look at all the licenses in this repository. https://github.com/olikraus/u8g2/blob/master/LICENSE, https://github.com/olikraus/u8g2/wiki/fntgrp

> An alternative for Zephyr OS I see going into the direction of "develop own graphic subsystem" which would be the nice and beautiful way:
>
>     * take u8g2's drivers and refactor them underneath [...]

No, do not take u8g2's drivers, there is no one even slightly worthy of not being rewritten. I am not convinced in any way.

> Company internal we have not decide yet whether to add a SH1107 driver (which ever it will be in the end) in to Zephyr OS and to continue to use CFB, as limited as it is or to switch to u8g2.

Looking at SH1107 controller datasheet, if it cannot be supported by `drivers/display/ssd1306.c` driver, which also supports the SH1106 controller, then it should be very easy to implement SH1107 driver based on it.

gmarull, 2023-05-30:
Closing, U8G2 can decide to support Zephyr by adding a `zephyr/module.yml` file and appropriate glue code.

closed 2023-05-30 by gmarull