zephyrproject-rtos/zephyr #38764: CBPRINTF_FP_SUPPORT does not work after NEWLIB_LIBC enabled

kclauah, 2021-09-23:
**Describe the bug**
Terminal unable to print float value if these two config enabled.
CONFIG_NEWLIB_LIBC=y 
CONFIG_CBPRINTF_FP_SUPPORT=y
If CBPRINTF_FP_SUPPORT is set only, it can correctly print float to terminal.
I found out this issue when I testing samples/sensor/lsm6dso on nrf52dk_nrf528322.
I think this issue is related to https://github.com/zephyrproject-rtos/zephyr/issues/37777

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project with these config.
CONFIG_CBPRINTF_FP_SUPPORT=y
CONFIG_NEWLIB_LIBC=y
2. Add the following code.
float test_float = 1.23;
printf("float: %f\n",test_float);
3. See the result on terminal

**Expected behavior**
Terminal print blank if CONFIG_NEWLIB_LIBC=y & CONFIG_CBPRINTF_FP_SUPPORT=y
float: 
It can correctly print float if remove CONFIG_NEWLIB_LIBC
float: 1.230000

**Impact**
I have to replaced all printf with printk in order to show float on terminal.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Windows 10 64-bit
 - Toolchain  Zephyr SDK 2.6.0.rc-1

**Additional context**

stephanosio, 2021-09-23:
When newlib is enabled, `printf` is implemented by the newlib (and not by the cbprintf).

Please make sure that `CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y`.

kclauah, 2021-09-23:
Thank you, your solution works.
I just found out that newlib has its own implementation of printf and enabling newlib override cbprintf.
Sorry for the dumb question.

stephanosio, 2021-09-23:
No problem. This can be quite confusing to the people who are new to Zephyr.

Closing since the question has been answered.

closed 2021-09-23 by stephanosio