zephyrproject-rtos/zephyr #38765: samples: create an OLED example

DhruvaG2000, 2021-09-23:
**Is your enhancement proposal related to a problem? Please describe.**
There is no existing example to help get started with OLED modules like SSD1306. An example that demonstrates the use of ``lvgl`` would also be really useful. 

**Describe the solution you'd like**
Write an example that will display "Hello World" on any display module connected via IIC to any board, since ```LVGL`` is a generic library/ module. 

**Describe alternatives you've considered**
Something similar to [u8g2 example](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/HelloWorld/HelloWorld.ino) can be provided within ``samples/zephyr``

**Additional context**
N/A.

jfischer-no, 2021-09-23:
We do not need samples for this specific display technology.
We have generic samples for cfb, lvgl: _samples/subsys/display_ and display controller driver API: _samples/drivers/display_.

https://docs.zephyrproject.org/latest/samples/subsys/display/lvgl/README.html

gmarull, 2021-09-23:
I agree that display samples don't need to be dependent on a particular technology, but rather on API/libraries. One could use the same to display content on devices made using LCD or OLED technologies.

Closing this issue, as there are existing samples that can be used (`samples/subsys/display/*`).

closed 2021-09-23 by gmarull