zephyrproject-rtos/zephyr #38782: CONFIG_BT_CTLR_DATA_LENGTH_MAX=250 causes pairing compatibility issues with many devices

mshaw11, 2021-09-23:
Having had many issues getting bluetooth pairing to work on a range of devices I have identified that setting `CONFIG_BT_CTLR_DATA_LENGTH_MAX=250` (the max value) was the root cause.

I had failed to pair to any device except for snapdragon based smartphones. 

**To Reproduce**
Extend bluetooth buffers to maximum:
`
CONFIG_BT_BUF_ACL_RX_SIZE=250
CONFIG_BT_CTLR_DATA_LENGTH_MAX=250
CONFIG_BT_L2CAP_TX_MTU=260
CONFIG_BT_RX_STACK_SIZE=2048
CONFIG_BT_BUF_ACL_RX_SIZE=251
CONFIG_BT_BUF_EVT_RX_SIZE=251
CONFIG_BT_BUF_CMD_TX_SIZE=251
CONFIG_BT_CONN_TX_MAX=10
CONFIG_BT_L2CAP_TX_BUF_COUNT=10
CONFIG_BT_L2CAP_TX_FRAG_COUNT=10
CONFIG_BT_BUF_ACL_TX_COUNT=10
CONFIG_BT_BUF_ACL_TX_SIZE=251
CONFIG_BT_BUF_CMD_TX_SIZE=251
CONFIG_BT_BUF_CMD_TX_COUNT=10
`

**Expected behavior**
One of three solutions:
1. Pairing should succeed. 
2. Maximum allowed value should be lowered.
3. There should be an issue on github people can easily find in a search to warn them ;)

**Impact**
Pairing subtly breaks for many connecting devices. This is an annoyance but also extremely difficult to debug because it's not obvious there is a connection between zephyr side buffer lengths and pairing functionality. Just annoying more than anything.

**Logs and console output**
Getting Pairing Failed callback with error code BT_HCI_ERR_CONN_TIMEOUT.

**Environment (please complete the following information):**
 - OS: MAC O/S
 - Toolchain gnuarmemb
- Release: zephyr-2.6.0
- Board nrf52840 in peripheral mode
- Fails connecting to: Various windows based laptops, various tablets. All devices not Snapdragon based (that I have seen).

**Additional context**
It works/bonds when I set CONFIG_BT_CTLR_DATA_LENGTH_MAX=27 and CONFIG_BT_BUF_ACL_RX_SIZE=251 so it seems limited to this one config option. When I reduce `CONFIG_BT_BUF_ACL_RX_SIZE` to 200 all works perfectly.

carlescufi, 2021-09-23:
@mshaw11 can you please use the [Nordic Sniffer](https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le) to capture some traces of the failing pairing procedures and upload them here? Additionally, please give us a way to reproduce the problem locally (eg. a list of phones that fail?)

mshaw11, 2021-09-23:
Hi Carles

The devices it fails in pairing to were almost all devices I tried except snapdragon based phones, so I had success with:
Google Pixel 4 - Android 11.
Xiaomi Mi - Android 11.
Samsung tablets x 2 - 1 Android 10, 1 Android 9

I failed to pair to:
Lenovo M8 tablet.
Multiple ubuntu based laptops including an Asus Zenbook 14, Dell Insprion 15 7510.
Multiple windows based laptops from an Alienware (model unknown), Dell XPS 15 and Zenbook 14.

I'm fairly confident if you add the config listed above to any nrf52840 zephyr peripheral sample you will reproduce the issue with most laptops with BLE support.

I unfortunately will not have the opportunity to retest as I will not have access to the equipment I was developing on any longer. This issue plagued me enough that I felt it worth raising the issue.

carlescufi, 2021-09-23:
>I'm fairly confident if you add the config listed above to any nrf52840 zephyr peripheral sample you will reproduce the issue with most laptops with BLE support.

Yep, I can reproduce the issue with your config and an iPhone 11.

cvinayak, 2021-09-24:
@mshaw11 and @carlescufi 

`CONFIG_BT_CTLR_DATA_LENGTH_MAX` is used in the Controller to configure, limit PDU reception size and to reduce RAM usage for the PDU receive buffers.

With 250 as the configured value, if the peer is sending a PDU with size greater than 250 because of ATT MTU Exchange with server receive MTU of 247 bytes (`CONFIG_BT_BUF_ACL_RX_SIZE=251` - 4), then the controller will drop the PDUs.

Please check if `CONFIG_BT_BUF_ACL_RX_SIZE=250` resolves the issue.

mshaw11, 2021-09-24:
Can confirm setting all the buffers listed above to 251 also fixes the issue.

carlescufi, 2021-09-24:
Thanks @cvinayak. Let’s close the issue then.

closed 2021-09-24 by carlescufi