zephyrproject-rtos/zephyr #38799: BLE central_ht only receives 7 notifcations

1hannah11, 2021-09-23:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/38760

<div type='discussions-op-text'>

<sup>Originally posted by **1hannah11** September 23, 2021</sup>
Hi, I've been using the Bluetooth Central_ht sample with a few changes (displaying data differently and changed uuids for my peripheral) and I only receive 7 "notifications" from the notify_func, although I am constantly sending data from the peripheral which is connected. after 7 notifications it doesn't disconnect or unsubscribe it just acts as if its not receiving any data. If i connect a different central to the peripheral I receive constant data as I should be. 

I am using a microbit V1 for the BLE peripheral and a Microbit V2 as the BLE central.  The code for the peripheral is .ino.

Does any one know why I am encountering this problem?

Changes I made to the Central_ht sample:

//Declaring Temp Service UUIDs
#define BT_UUID_TEMP_SERVICE_VAL 0xccc0  //declares uuid
#define BT_UUID_TEMP_SERVICE \
	BT_UUID_DECLARE_16(BT_UUID_TEMP_SERVICE_VAL) //creates 16 bit uuid
//Temperature reading characterisatic uuid
#define BT_UUID_CHARACTERISTIC_VAL 0xccc1
#define BT_UUID_TEMP_READING\
	BT_UUID_DECLARE_16(BT_UUID_CHARACTERISTIC_VAL)

And:
static uint8_t notify_func(struct bt_conn *conn,
			   struct bt_gatt_subscribe_params *params,
			   const void *data, uint16_t length)
{

	if (!data) {
		printk("[UNSUBSCRIBED]\n");
		params->value_handle = 0U;
		return BT_GATT_ITER_STOP;
	}

	float rawData = ((float *)data)[0];
	double temperature = (float)rawData;

	printf("Temperature %gC.\n", temperature);

	return BT_GATT_ITER_CONTINUE;
}
</div>

carlescufi, 2021-09-27:
@1hannah11 

> it just acts as if its not receiving any data.

Can you enable `CONFIG_BT_DEBUG_GATT` and then post the output you get in the central please?

carlescufi, 2021-09-27:
I just tested the vanilla sample and works fine. This must be on your code. I suggest you ask in Discord or go back to the original GitHub discussion because this doesn't look like a bug in Zephyr.

closed 2021-09-27 by carlescufi

1hannah11, 2021-09-28:
Sorry, yes you are right it was my peripheral code. Used the peripheral_ht sample and all was normal. Sorry for any inconveniences