zephyrproject-rtos/zephyr #38831: CPP stack overflow in build v2.6.99-ncs1

hakkimakki, 2021-09-24:
**Describe the bug**
When I compile my project [link](https://github.com/hakkimakki/NRD/tree/main/test), i get no errors when compiling it either with the v2.6.99-ncs1 (nrf-connect-sdk v1.7.0) or v2.6.0-rc1-ncs1 (nrf-connect-sdk v1.6.1). However when i execute it compiled with v2.6.0-rc1-ncs1 (nrf-connect-sdk v1.6.1) it works fine. When compiled with v2.6.99-ncs1 (nrf-connect-sdk v1.7.0) a stack overflow occurs. The overflow happens at calling the ctor (see log).

**To Reproduce**
Steps to reproduce the behavior:
1. get the projekt _test_ [link](https://github.com/hakkimakki/NRD/tree/main/test)
2. place a copy of _test_ inside the two workspaces root folders (v2.6.99... and v2.6.0...)
3. inside both projects run:
4. west build -p auto -b nrf52840dk_nrf52840
5. west flash
6. See error

**Expected behavior**
As it is not clear to me why the overflow happens. Could be a basic mistake or a change in library of cpp.

**Impact**
The bug keeps me from updating to the new version.

**Logs and console output**
*** Booting Zephyr OS build v2.6.0-rc1-ncs1  ***
Started
Landys+Gyr E450 Push Listener started
Packet Complete: 1
More Data: 1
Packet Complete: 1
More Data: 1
Packet Complete: 1
More Data: 0
GXDLMSPushSetup 0.8.25.9.0.255 1: 0.8.25.9.0.255
GXDLMSData 0.0.96.1.0.255 2: 35 36 35 35 32 37 37 33
GXDLMSRegister 1.0.1.7.0.255 2: 71
GXDLMSRegister 1.0.2.7.0.255 2: 0
GXDLMSRegister 1.1.1.8.0.255 2: 214002
GXDLMSRegister 1.1.2.8.0.255 2: 0
GXDLMSRegister 1.1.5.8.0.255 2: 8349
GXDLMSRegister 1.1.6.8.0.255 2: 0
GXDLMSRegister 1.1.7.8.0.255 2: 0
GXDLMSRegister 1.1.8.8.0.255 2: 106530
GXDLMSRegister 1.0.31.7.0.255 2: 36
GXDLMSRegister 1.0.51.7.0.255 2: 17
GXDLMSRegister 1.0.71.7.0.255 2: 16
Server address: 1 Client Address: 13311
Done

*** Booting Zephyr OS build v2.6.99-ncs1  ***
Started
Landys+Gyr E450 Push Listener started
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x20002958
E: r0/a1:  0x20000138  r1/a2:  0x2000013c  r2/a3:  0x200021e0
E: r3/a4:  0x0003079b r12/ip:  0x00000000 r14/lr:  0x00026c27
E:  xpsr:  0x20000000
E: Faulting instruction address (r15/pc): 0x20000138
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x20000488 (main)
E: Halting system

**Environment (please complete the following information):**
 - OS: Windows 10 
 - Toolchain: GNU Arm Embedded Toolchain 9_2019-q4-major

**Additional context**
When debbuging the overflow happens right after stepping over this line:
`CGXReplyData data;`
However any further breakpoint inside of the ctor is not reached.

stephanosio, 2021-09-25:
It does not look like the repository you linked above is publicly available, so we have no way of reproducing this.

It is very likely that you are actually running out of stack (hence stack overflow), so try increasing stack size.

hakkimakki, 2021-09-25:
Sorry the repo is now public. 

I have already increased the stack from 8192 to 16000 but still the same.

closed 2021-12-09 by github-actions[bot]