zephyrproject-rtos/zephyr #38836: Build system should allow some resizing between linker passes 1 and 2

mbolivar-nordic, 2021-09-24:
**Is your enhancement proposal related to a problem? Please describe.**

This is meant to track the 'real fix' to follow along the band-aid in #38834.

The userspace implementation builds a hash table using kernel object addresses as keys. These addresses must therefore remain valid between linker passes. Up until now, the solution to meeting this requirement has been that we cannot resize any values between passes 1 and 2.

However, this is causing pain and bugs. The pain is that unnecessary ROM is wasted by gen_handles.py (see #38834 and associated commit logs for more details) -- even when `CONFIG_USERSPACE=n`. An example bug is that merging #37836 led to #38181 because we ran out of room in an array that logically should have been resized between passes. In general, this is a restriction that is really starting to chafe where device handles are concerned.

**Describe the solution you'd like**

After discussing a bit with @tejlmand, it seems potentially possible to meet the userspace requirements *and* be able to resize our handles arrays if the linker scripts only ever allocate device handle arrays *after* all of the statically allocated kobjects. Such kobjects could be placed by their allocation macros into locations that come earlier in the image's ROM than the memory for device handle arrays. In this way, even if the handle arrays are resized, the addresses of the kernel objects should remain the same after the second pass link.

**Describe alternatives you've considered**

There is a wider-ranging RFC in https://github.com/zephyrproject-rtos/zephyr/issues/32129. That seems like potential overkill given that nothing else currently really "needs" to be resized between linker passes, assuming there's no showstopper associated with this issue.

nashif, 2021-09-29:
@dcpleung FYI

JordanYates, 2021-10-07:
I think the solution involving manually placing objects in ROM is unnecessarily fragile, but I also see the point that #32129 is overkill.

I am thinking of one additional linker step/elf file which is explicitly for scripts that require and produce size invariance.
```
zephyr_prebuilt.elf -> zephyr_fixed_layout.elf -> zephyr.elf
```
Where scripts such as `gen_handles.py` run on `zephyr_prebuilt.elf`, while `gen_kobject_list.py` and friends run on `zephyr_fixed_layout.elf`.

ceolin, 2021-10-15:
> I think the current `zephyr_prebuilt.elf` is serving this purpose?

the problem is that both `gen_kobject_list.py` and `gen_handles.py` operate in this stage and the latter should be able to resizing some things, which consequently breaks the former :/ And additional step is needed, or properly put the kobjects before the objects that need to be resized, but I also think that this is fragile.

JordanYates, 2021-10-19:
I had a bit of a discussion with @carlescufi and worked out that Zephyr actually already uses 3 linker stages when `CONFIG_USERSPACE` is enabled (`app_smem_unaligned_prebuilt.elf`). So another way of going about this (instead of an additional stage between `zephyr_prebuilt.elf` and `zephyr.elf`) is to generate `app_smem_unaligned_prebuilt.elf` under additional conditions (`CONFIG_HAS_DTS`). `gen_handles.py` can then run on `app_smem_unaligned_prebuilt.elf`, generating the correct arrays in time for `zephyr_prebuilt.elf`.

commit 8d93217a (2021-11-16): cmake: new global property for generated files

Introduce a new global property for source files that are generated from
the initial Zephyr link (app_smem_unaligned_prebuilt). This source list
is used for files which will introduce address shifts into the final
binary, which need to be present in `zephyr_prebuilt.elf` for
`CONFIG_USERSPACE` scripts to correctly generate `zephyr.elf`.

This resolves #38836.
closed 2021-11-16 by carlescufi