zephyrproject-rtos/zephyr #38840: Blinky example on MIMXRT1050EVKB hangs up in rom code when compiled for QSPI flash.

mmurty, 2021-09-25:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/38839

<div type='discussions-op-text'>

<sup>Originally posted by **mmurty** September 25, 2021</sup>
Hi Everyone,
I am trying to get started with zephyr + NXP's MIMXRT1050EVKB board.
I tested the blinky example with MIMXRT1020EVK board and it worked fine. 
Now I am trying to build the same blinky example for MIMXRT1050EVKB board for booting from QSPI flash but I see that the code hangs up in the ROM code as soon as the binary is flashed. 

Below is the compilation log:
---------------------------------------------------------------------------------------------------------------------
west build -p auto -b mimxrt1050_evk_qspi samples/basic/blinky
-- west build: making build dir /home/mahesh/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/mahesh/zephyrproject/zephyr/cmake/app/boilerplate.cmake
CMake Deprecation Warning at /home/mahesh/zephyrproject/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
  The OLD behavior for policy CMP0079 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  /home/mahesh/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/mahesh/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Application: /home/mahesh/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.7.0-rc1 (/home/mahesh/zephyrproject/zephyr), build: v2.7.0-rc1-164-gca4c71ca0262
-- Found Python3: /home/mahesh/zephyrproject/.venv/bin/python3 (found suitable exact version "3.8.10") found components: Interpreter
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: mimxrt1050_evk_qspi
-- Cache files will be written to: /home/mahesh/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.13.0 (/home/mahesh/zephyr-sdk-0.13.0)
-- Found dtc: /home/mahesh/zephyr-sdk-0.13.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/mahesh/zephyrproject/zephyr/boards/arm/mimxrt1050_evk/mimxrt1050_evk_qspi.dts
-- Generated zephyr.dts: /home/mahesh/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/mahesh/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/mahesh/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/mahesh/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/mahesh/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/mahesh/zephyrproject/zephyr/boards/arm/mimxrt1050_evk/mimxrt1050_evk_qspi_defconfig'
Merged configuration '/home/mahesh/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/mahesh/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/mahesh/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mahesh/zephyr-sdk-0.13.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mahesh/zephyrproject/zephyr/build
-- west build: building application
[1/143] Preparing syscall dependency handling

[136/143] Linking C executable zephyr/zephyr_prebuilt.elf

[143/143] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           OCRAM:          0 GB       256 KB      0.00%
           FLASH:       22856 B         8 MB      0.27%
            SRAM:        3968 B        32 MB      0.01%
            ITCM:          0 GB       128 KB      0.00%
            DTCM:          0 GB       128 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
---------------------------------------------------------------------------------------------------------------------
I am using JLink to flash and debug.
I must be doing some basic stuff wrong as I am new to zephyr. Hence I am unsure whether this is an issue.
Please take a look at let me know your findings.</div>

mmahadevan108, 2022-09-14:
@mmurty I think this is fixed in the current Zephyr main. Please let us know.

mmahadevan108, 2024-02-13:
@mmurty The config for QSPI flash is now part of the HAL. The file is called `evkbimxrt1050_flexspi_nor_qspi_config.c`.