zephyrproject-rtos/zephyr #38846: net: dns-sd: additional rfc6733 features

cfriedt, 2021-09-25:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, dns-sd does not support
1. domain enumeration (chapter 11)
2. sub-types (chapter 7.2)
3. service domains (chapter 7.2)

Items 2 and 3 deal with fully-qualified names (FQN's). Examples of FQN's that are unsupported are of the form:

```
<sub>._sub.<sn>._tcp.<servicedomain>.<parentdomain>.
```
or anything other than "local" for a `<domain>`, where
```
<domain> := <servicedomain>.<parentdomain>.
```

The `struct dns_sd_rec` actually has insufficient fields to represent all types of names.

**Describe the solution you'd like**
* 1 additional local "service domain", either configured from e.g. DHCP or via Kconfig
* remove the `proto`, `service`, and `instance` fields of `struct dns_sd_rec`
* use only a `name` field and `name_size`, `domain` field and `domain_size`, and `text` and `text_size`
* use accessor methods to extract the above values into callee-provided storage
* support domain enumeration via kconfig switch (ch 11)
* support all fully-qualified names described in ch 7.2
* an additional pass of helpful features from RFC 6763

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
https://datatracker.ietf.org/doc/html/rfc6763

AllemanMa, 2021-12-20:
The driver seems to not initiate device-side announcements or probing at startup as described in section 8: https://datatracker.ietf.org/doc/html/rfc6762#section-8. This causes my device to remain undiscovered when the server has avahi-daemon running.

cfriedt, 2021-12-21:
> The driver seems to not initiate device-side announcements or probing at startup as described in section 8: https://datatracker.ietf.org/doc/html/rfc6762#section-8. This causes my device to remain undiscovered when the server has avahi-daemon running.

Thanks for finding this! I see that it's part of the greater mdns picture (rfc6762) so I might create a separate issue for general mdns improvements.

cfriedt, 2022-08-10:
Pushing this back, as I have no actual dedicated hours to work on this at the moment (most of my upstream work is still unfortunately evenings and weekends).

cfriedt, 2023-04-19:
@pdgendt - in case you have bandwidth to make additional improvements. Feel free to ping me on Discord for any clarification. In particular, accessor methods would be the preferred way to extract fields.

pdgendt, 2023-04-20:
> @pdgendt - in case you have bandwidth to make additional improvements. Feel free to ping me on Discord for any clarification. In particular, accessor methods would be the preferred way to extract fields.

@cfriedt thanks, it could be something for the long run though. We're currently more focused on RFC6762, and filling some gaps there.