zephyrproject-rtos/zephyr #38857: Bluetooth: Host: Clear up confusing `bt_conn` pointer namings

Thalley, 2021-09-27:
**Is your enhancement proposal related to a problem? Please describe.**
The `struct bt_conn` struct is used for ACL, SCO, BR and ISO connections, but is still most often just used as 
`(struct bt_conn *conn)` in functions, where it isn't clear what type of connection it really is. 

**Describe the solution you'd like**
It has been done for `sco` mostly where the functions have `struct bt_conn *sco_conn` as the argument to the functions, making it clear that it is a `sco` connection. The solution proposed here to is to spread that convention to `le`, `br` and `iso` as well. 

**Describe alternatives you've considered**
It is possible that the `struct bt_conn` should refactored as a whole, as some of the information in the struct doesn't apply to all the types of connections, e.g. an ISO connection does not have any L2CAP channels

**Additional context**
Some functions, e.g. `bt_conn_ref` work for all types of connections, and shouldn't be updated. 

It is trivial to change this in the internal functions that does not affect the API, and it is suggested to start with that. Doing a complete renaming in the public API requires more effort.

Thalley, 2024-02-09:
I would still consider this valid - It is still a continuous source of confusion (especially now that BREDR is going alive again).

I also fully expect that there are multiple functions that take a `bt_conn` that does not have a type check (e.g. what happens if you do `bt_gatt_discover` on an ISO `bt_conn`?)

jori-nordic, 2024-02-09:
> I also fully expect that there are multiple functions that take a bt_conn that does not have a type check (e.g. what happens if you do bt_gatt_discover on an ISO bt_conn?)

I believe this is the real issue, and could be addressed in a non-breaking way.

If you're only proposing a param name change, that should also not break existing apps, correct me if I'm wrong though.

If you're rather proposing a refactor of `bt_conn`, e.g. that you call `bt_gatt_notify(struct bt_conn_le *conn)` then that's another can of worms, and also debatable if the fix is worth the breakage.

Thalley, 2024-02-09:
> > I also fully expect that there are multiple functions that take a bt_conn that does not have a type check (e.g. what happens if you do bt_gatt_discover on an ISO bt_conn?)
> 
> I believe this is the real issue, and could be addressed in a non-breaking way by having a common `is_le() -> return EINVAL` check on the LE APIs.
> 
> If you're only proposing a param name change, that should also not break existing apps, correct me if I'm wrong though.
> 
> If you're rather proposing a refactor of `bt_conn`, e.g. that you call `bt_gatt_notify(struct bt_conn_le *conn)` then that's another can of worms, and also debatable if the fix is worth the breakage.

Non-breaking changes would likely be best. 
That being said, it may be an overall better solution to use more specific types in the API, if only a subset of types can be used, but that me less useful when we also have functions like `bt_conn_disconnect` that work on multiple types. 

So perhaps just the name change (originally suggested by this issue) + documentation for conn type + type checks would be a nice and non-breaking change.

Thalley, 2025-02-06:
Coverity found an issue this week related to this issue: https://scan9.scan.coverity.com/#/project-view/43529/12996?selectedIssue=489215

Basically, even if used in a GATT callback context `bt_conn_index(conn)`, in the eyes of coverity, is no longer restricted to `0 to CONFIG_BT_MAX_CONN - 1`, as `bt_conn_index` could also return indexes of ISO or SCO connections.