zephyrproject-rtos/zephyr #38861: g0b1re: Garbage uart data when running `spi_flash_at45` sample with DMA enabled

ycsin, 2021-09-27:
**Describe the bug**
I'm trying the `spi_flash_at45` on `nucleo_g0b1re`, and found that there are some garbage data in the debug log when SPI DMA is enabled.

**To Reproduce**

```shell
west build -b nucleo_g0b1re --pristine samples\drivers\spi_flash_at45
west flash
```

nucleo_g0b1re.dts:
```dts
&dma1 {
	status = "okay";
};

&dmamux1 {
	status = "okay";
};

&spi1 {
	pinctrl-0 = <&spi1_nss_pb0 &spi1_sck_pa5
		     &spi1_miso_pa6 &spi1_mosi_pa7>;
	status = "okay";
	dmas = <&dmamux1 2 17 0x20440
		&dmamux1 1 16 0x20480>;
	dma-names = "tx", "rx";
	cs-gpios = <&gpioc 8 GPIO_ACTIVE_LOW>;
	at45db0: at45db041e@0 {
		compatible = "atmel,at45";
		reg = <0>;
		spi-max-frequency = <15000000>;
		label = "DATAFLASH_0";
		jedec-id = [1f 27 01];
		size = <33554432>;
		sector-size = <65536>;
		block-size = <4096>;
		page-size = <512>;
		use-udpd;
		enter-dpd-delay = <4000>;
		exit-dpd-delay = <180000>;
		status = "okay";
	};
};
```

prj.conf:
```conf
# Not related, just for testing
CONFIG_SHELL=y

# Extra DMA configs for the sample
CONFIG_DMA=y
CONFIG_SPI_STM32_DMA=y
CONFIG_SPI_STM32_INTERRUPT=n

# CONFIG_SPI_ASYNC=y
```

**Expected behavior**
UART with/without DMA should behaves the same

**Impact**
Nothing major for a test application, can cause error if the garbage appears in uart communication with an external modem.

**Logs and console output**
With DMA:
```
*** Booting Zephyr OS build v2.7.0-rc2-134-gb3be5ba11ce2  ***
DataFlash sample on nucleo_g0b1re
Using DATAFLASH_0, chip size: 4194304 bytes (page: 512)
Reading the first byte of the test region ... OK
Preparing test content starting with 0x3A.
Writing the first half of the test region...

K
Writing the second half of the test region... 1;32muaOK
Reading the whole test region... rt:~$ OK
Checking the read content... OK
Putting the flash device into suspended state... OK
```

Without DMA:
```
*** Booting Zephyr OS build v2.7.0-rc2-134-gb3be5ba11ce2  ***
DataFlash sample on nucleo_g0b1re
Using DATAFLASH_0, chip size: 4194304 bytes (page: 512)
Reading the first byte of the test region ... OK
Preparing test content starting with 0x3B.
Writing the first half of the test region... OK
Writing the second half of the test region... OK
Reading the whole test region... OK
Checking the read content... OK
Putting the flash device into suspended state... OK
```

**Environment (please complete the following information):**
 - Win 10
 - gnuarmemb
 - 611284021c

**Additional context**
Add any other context about the problem here.

ycsin, 2021-09-27:
@erwango I have tried the old `uart_stm32.c` code and the latest one, both results are identical.

erwango, 2021-09-27:
@ycsin Have you tried to set dma prio to the same level as spi one ?

ycsin, 2021-09-27:
@erwango Sorry, I don't really understand. I only enabled DMA for SPI in the devicetree and left UARTs untouched, is there any dma prio that I should configure?

erwango, 2021-09-27:
Try to set interrupts prio to 5 (same as SPI), here:
```
		dma1: dma@40020000 {
			compatible = "st,stm32-dma-v2";
			#dma-cells = <3>;
			reg = <0x40020000 0x400>;
-			interrupts = <9 0 10 0 10 0 11 0 11 0>;
+			interrupts = <9 5 10 5 10 5 11 5 11 5>;
			clocks = <&rcc STM32_CLOCK_BUS_AHB1 0x1>;
			dma-requests = <5>;
			dma-offset = <0>;
			status = "disabled";
			label = "DMA_1";
		};
```

ycsin, 2021-09-27:
It is overwritten in g0b1.dtsi and has different length:

```
interrupts = <9 0 10 0 10 0 11 0 11 0 11 0 11 0>;
```

erwango, 2021-09-27:
> > Try to set interrupts prio to 5 (same as SPI), here:
> 
> It is overwritten in g051.dtsi and has different length:
> 
> ```
> interrupts = <9 0 10 0 10 0 11 0 11 0 11 0 11 0>;
> ```
> 
> What should I set this to?

These go by pairs. In first pair '9 0', '9' is the IRQ line, '0' is the prio.
Replace all '0' by '5'.

ycsin, 2021-09-27:
> These go by pairs. In first pair '9 0', '9' is the IRQ line, '0' is the prio.
> Replace all '0' by '5'.

I overwrite it in the `nucleo_g0b1re.dts` now

```dts
&dma1 {
	interrupts = <9 5 10 5 10 5 11 5 11 5 11 5 11 5>;
	status = "okay";
};

&dmamux1 {
	status = "okay";
};

&spi1 {
	pinctrl-0 = <&spi1_nss_pb0 &spi1_sck_pa5
		     &spi1_miso_pa6 &spi1_mosi_pa7>;
	status = "okay";
	dmas = <&dmamux1 2 17 0x20440
		&dmamux1 1 16 0x20480>;
	dma-names = "tx", "rx";
	cs-gpios = <&gpioc 8 GPIO_ACTIVE_LOW>;
	at45db0: at45db041e@0 {
		compatible = "atmel,at45";
		reg = <0>;
		spi-max-frequency = <15000000>;
		label = "DATAFLASH_0";
		jedec-id = [1f 27 01];
		size = <33554432>;
		sector-size = <65536>;
		block-size = <4096>;
		page-size = <512>;
		use-udpd;
		enter-dpd-delay = <4000>;
		exit-dpd-delay = <180000>;
		status = "okay";
	};
};
```

But unable to compile:
```
C:/Users/yongc/zephyrproject/zephyr/drivers/dma/dma_stm32.c: In function 'dma_stm32_config_irq_0':
../../../../include/toolchain/gcc.h:61:36: error: static assertion failed: "Invalid interrupt priority. Values must not exceed IRQ_PRIO_LOWEST"
   61 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
      |                                    ^~~~~~~~~~~~~~
../../../../include/arch/arm/aarch32/irq.h:97:2: note: in expansion of macro 'BUILD_ASSERT'
   97 |  BUILD_ASSERT((flags_p & IRQ_ZERO_LATENCY) || \
```

erwango, 2021-09-27:
Ok, my bad, it seems that 5 is not a valid value on GO and besides SPI prio is 0 on G0.
So I guess my suggestion won't work.
Anyway, you can try to set prio to '4' and see if things get better, but I''m not so sure now.

ycsin, 2021-09-27:
> Try to set interrupts prio to 5 (same as SPI)

Is this written anywhere?
```dts
		spi1: spi@40013000 {
			compatible = "st,stm32-spi-fifo", "st,stm32-spi";
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40013000 0x400>;
			clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00001000>;
			interrupts = <25 0>;
			status = "disabled";
			label = "SPI_1";
		};
```

ycsin, 2021-09-27:
> Anyway, you can try to set prio to '4' and see if things get better, but I''m not so sure now.

I've tried 4 but it also fails to compile, 1,2,3 compiles, but same result

```
DataFlash sample on nucleo_g0b1re
Using DATAFLASH_0, chip size: 4194304 bytes (page: 512)
Reading the first byte of the test region ... OK
Preparing test content starting with 0x41.
Writing the first half of the test region...

K
Writing the second half of the test region... 1;32muaOK
Reading the whole test region... rt:~$ OK
Checking the read content... OK
Putting the flash device into suspended state... OK
```

erwango, 2021-09-27:
> I've tried 4 but it also fails to compile, 1,2,3 compiles, but same result
>
> ```
> DataFlash sample on nucleo_g0b1re
> Using DATAFLASH_0, chip size: 4194304 bytes (page: 512)
> Reading the first byte of [...]

Ok, Thanks for having had a try. We'll need to investigate a bit more then.

closed 2021-12-11 by github-actions[bot]