zephyrproject-rtos/zephyr #38877: Running the zephyr elf natively on an arm a53 machine (ThunderX2) with KVM emulation

AshwinKrishn, 2021-09-27:
**Is your feature request related to a problem? Please describe.**
I am trying to boot the elf generated on a aarch64 machine. It seems to boot without emulation using zephyr SDK's qemu-system-aarch64. But it does not boot with system's qemu-aarch64. The KVM emulation is not seeming to work. IS there a reason why emulation booting is not enabled ? 

**Describe the solution you'd like**
I would like to boot zephyr.elf of my application on an aarch64 machine with kvm enabled. 

**Describe alternatives you've considered**
I tried on two different aarch64 machines , but nothing seems to allow booting.  we get the following errors. 
==============
$/home/ashwin/zephyr-sdk-0.13.0/sysroots/aarch64-pokysdk-linux/usr/bin/qemu-system-aarch64 --enable-kvm  -cpu cortex-a53 -nographic -machine virt,gic-version=3 -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -kernel /home/ashwin/zephyr.elf
**ERROR:**
qemu-system-aarch64: invalid accelerator kvm

=========
sudo qemu-system-aarch64  -cpu cortex-a53 -nographic -enable-kvm  -machine virt,secure=on,gic-version=3 -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -kernel zephyr.elf
**ERROR:**
kvm_init_vcpu failed: Invalid argument

**Additional context**
Enabling booting of the system on a native machine thorough qemu will expand integration of distributed emulation. Example: popcornlinux.org

AshwinKrishn, 2021-09-29:
This was an issue with QEMU. Build QEMU yourself which is a 3 hours of work. But it works.

closed 2021-09-29 by AshwinKrishn

DedEnginr, 2024-05-29:
> This was an issue with QEMU. Build QEMU yourself which is a 3 hour work. But it works.

Hi, 
I'm trying to boot zephyr in KVM in a linux system. Can you explain how you did it.
I tried it with the .elf, .efi, .bin files but nothing seems working.
I hope to get some lead here.