zephyrproject-rtos/zephyr #38893: g0b1re + spi_flash_at45 + flash_shell: First write always fails with `CONFIG_PM_DEVICE`

ycsin, 2021-09-28:
**Describe the bug**
I'm using nucleo g0b1re, testing spi_flash_at45 with default config + `CONFIG_SHELL=y`. I tried to write to some random offset of the AT45DB321E flash, but the first write always fails, but the subsequent write to the same or other offset is successful. If I issue a read first before a write, then the read & write will both be successful. I tried a few combinations (1, 2 & 3).

I then tried to remove the `use-udpd` from the dts, but then the whole program hang, see 4, but the read works, see 5. The sample's read/write works though, so I suspect something isn't right in the `flash write` command?

After that I removed the `use-udpd`, `enter-dpd-delay`, `exit-dpd-delay` & `CONFIG_PM_DEVICE=y`, then the first write starts to work, see 6.

**To Reproduce**
Build and run `spi_flash_at45` sample on nucleo_g0b1re + AT45DB321E with `CONFIG_SHELL=y`

**Expected behavior**
Read and write should work, with or without PM.

**Impact**
I have a new board with AT45DB321E, this might cause issue in my application, I haven't test it yet though.

**Logs and console output**

1. [write_write_read.txt](https://github.com/zephyrproject-rtos/zephyr/files/7240343/write_write_read.txt)
2. [read_write_read.txt](https://github.com/zephyrproject-rtos/zephyr/files/7240345/read_write_read.txt)
3. [write_read_write_read.txt](https://github.com/zephyrproject-rtos/zephyr/files/7240346/write_read_write_read.txt)
4. [write_without_use-udpd.txt](https://github.com/zephyrproject-rtos/zephyr/files/7240351/write_without_use-udpd.txt)
5. [read_write_without_use-udpd.txt](https://github.com/zephyrproject-rtos/zephyr/files/7240377/read_write_without_use-udpd.txt)
6. [write_read_without-PM.txt](https://github.com/zephyrproject-rtos/zephyr/files/7240352/write_read_without-PM.txt)

**Environment (please complete the following information):**
 - WIN10
 - GNUARMEMB
 - 611284021c

erwango, 2021-09-28:
> After that I removed the use-udpd, enter-dpd-delay, exit-dpd-delay & CONFIG_PM_DEVICE=y from my dts and the sample's prj.conf, then the first write starts to work, see 6.

From the description, the effect of `CONFIG_PM_DEVICE=y` is not clear. Are you able to see the issue when `CONFIG_PM_DEVICE=n` ?
I think isolating the issue in only one subsystem would make things easier to analyse.

ycsin, 2021-10-05:
I tried it again today and found that there's no error this time.

closed 2021-10-05 by ycsin