zephyrproject-rtos/zephyr #38898: hawkbit: Roadmap

ycsin, 2021-09-28:
This issue serve as a tracker for anyone interested in improving the Hawkbit subsys.
Information about Hawkbit server can be found at the Eclipse's Hawkbit [website](https://www.eclipse.org/hawkbit/).

- [ ] Weak-linked device attribute function(s), so that user can override it in application without patching the source - #37640
- [ ] Offload `flash_omg_buffer_write` to a work ([comment](https://github.com/zephyrproject-rtos/zephyr/issues/37324#issuecomment-919060806)) - #37324
- [ ] Currently the thread invoking `hawkbit` APIs require stack size of ~3kiB from my testing. We can reduce the RAM usage by using `k_malloc`, similar to the [updatehub](https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/mgmt/updatehub).
- [ ] Convert to use `setting` subsys from raw `NVS` R/W
- [ ] Consider erasing the entire image slot before the start of download to improve download performance.
- [ ] Resume-able download

Comments are welcomed.

real-tintin, 2023-11-13:
Hi,

Any updates w.r.t the Hawkbit roadmap?

We are using Hawkbit for our use case and have many times scratched our heads given the current implementation. I find that the implementation is rather difficult to follow (nested and complex execution/logic paths) and contains bugs e.g., no means to handle lost connections (has caused use troubled during ongoing downloads).

We have made custom adaptions to get it working for us. But we have now realised that a rewrite of the implementation would be needed, as we aim to add more functionality to it e.g., support for soft and force updates. I think that a state machine drive design i.e., a FSM would be well suited.

Any thoughts are welcome.

ycsin, 2023-11-14:
Hi there, thanks for your interest in hawkBit!

> Any updates w.r.t the Hawkbit roadmap?

Unfortunately no, I don't really have much time for this since I left the company that focused on IoT about 2 years ago.

> We are using Hawkbit for our use case and have many times scratched our heads given the current implementation. I find that the implementation is rather difficult to follow (nested and complex [...]

It is indeed intimidating at first glance, to follow the request and response it requires developers to literally scroll through the entire source. Download has to be restarted when the connection is lost 🙁, though it shouldn't be _too_ hard to implement resumable download, we _just_ have to save the progress into the nvs and modify the request string.

> e.g., support for soft and force updates. I think that a state machine drive design i.e., a FSM would be well suited.

Although not clearly defined, hawkBit currently works in a FSM-ish fashion, there's definitely a big room for improvements, or even be rewritten to use [SMF](https://docs.zephyrproject.org/latest/services/smf/index.html).

I also think that it is possible for hawkBit to do more than firmware upgrade, i.e. downloading custom payload & configuration files for the [Settings subsystem](https://docs.zephyrproject.org/latest/services/settings/index.html).

> But we have now realised that a rewrite of the implementation would be needed, as we aim to add more functionality to it

The `hawkBit` subsystem is currently not maintained, feel free to submit PRs!

maass-hamburg, 2024-04-05:
as there are currently a lot major hawkBit PRs open and we are all are having limited resources, I made a litte priority list, beginning with the most important IMO:

- [ ] #68806 (includes migration to settings api)
- [ ] #69293
- [ ] #70386
- [ ] #70207
- [ ] #71037

I did not include the already approved PRs and trivial PRs. If you have any thoughts about it, feel free to comment.