zephyrproject-rtos/zephyr #38900: support a generic gpio-key driver

ck-telecom, 2021-09-28:
## Introduction
Introducing a generic gpio-key driver

### Problem description

Every project I need rewrite a button driver to support button keys, so is there a gpio-key driver similar to Linux gpio-key.c?(sorry I used to work for Linux Driver and Kernel)

### Proposed change

Not sure the kscan is suitable for this generic gpio-key driver, so maybe introduce another type driver?

## Detailed RFC

gpio device tree definition like bellow:
	buttons {
		compatible = "gpio-keys";
		key_in: button_0 {
			gpios = <&gpio0 13 GPIO_ACTIVE_HIGH>;
			label = "Key in";
                        key_code = <KEY_XXX>; // both standard KEY and custom KEY support
		};
	};

with a generic gpio-key driver for application(not support long pressed detection)

### Proposed change (Detailed)
get dts property and set gpio configuration, ISR invoke handler

### Dependencies
standard gpio-key device tree

### Concerns and Unresolved Questions

not sure it is a good idea

## Alternatives

None

ck-telecom, 2021-09-28:
/*
 * SPDX-License-Identifier: Apache-2.0
 */
#define DT_DRV_COMPAT gpio_keys

#include <zephyr.h>
#include <device.h>
#include <drivers/gpio.h>

    DT_GPIO_BY_IDX();
    DT_INST_GPIO_FLAGS(DT_NODE)

struct gpio_keys_data {
//TBD
};

struct gpio_keys_config {
    struct gpio_dt_spec *spec;
    char *key_val;
    int num;
};

static struct gpio_callback gpio_keys_cb;

static void gpio_keys_isr(const struct device *dev, struct gpio_callback *cb, uint32_t pins)
{
    /* FIXME: we don't make sure each pin has the same pin number */
    struct gpio_keys_config *cfg = dev->config;
    int state = gpio_pin_get_dt(cfg->spec[pins]);
    /* how to get pin number & key_val? */
}

static int gpio_keys_init()
{
    int ret = 0;
    struct gpio_keys_config *cfg = dev->config;

    for (int i = 0; i < cfg->num; i++) {
        ret = gpio_pin_configure_dt(&cfg->spec[i], GPIO_INPUT);
        if (ret) {
            LOG_ERR("error %d gpio_pin_configure_dt", ret);
            break;
        }

        ret = gpio_pin_interrupt_configure_dt(&cfg->spec[i], GPIO_INT_EDGE_BOTH);
        if (ret) {
            LOG_ERR("error %d gpio_pin_interrupt_configure_dt", ret);
            break;
        }
        /* FIXME: in case if different pin number is same? */
        gpio_init_callback(&gpio_keys_cb, gpio_keys_isr, BIT(cfg->spec[i].pin));
        ret = gpio_add_callback(&cfg->spec[i].port, &gpio_keys_cb);
        if (ret) {
            LOG_ERR("error %d gpio_add_callback", ret);
            break;
        }
    }

    return ret;
}

struct gpio_keys_config config = {
    .spec = GPIO_DT_SPEC_GET_OR() /* how to get all sw0 - swn ??? or in gpio_keys_data */
    .num = DT_NUM_NODES(buttons);
};

carlescufi, 2021-09-28:
@superna9999 @gmarull @mbolivar-nordic @galak FYI

henrikbrixandersen, 2021-09-28:
I am not sure I understand the benefit of adding yet another driver layer on top of the already existing GPIO interrupt/callback mechanism? What is the problem with the existing GPIO API based solution?

superna9999, 2021-09-28:
it could be useful if you had a proper key event mapping & handling library alongside, as-is it's nothing more than a gpio interrupt sample code

gmarull, 2021-09-28:
Regarding the question `/* how to get all sw0 - swn ??? or in gpio_keys_data */`, you can obtain all keys like this:

```c
struct gpio_keys_config {
    struct gpio_dt_spec spec;
    const char *key_val;
    int num;
};

#define INIT_KEY_CONFIG(node_id) \
    { .spec = GPIO_DT_SPEC_GET(node_id, gpios), \
      .key_val = DT_LABEL(node_id) \
      .num = DT_PROP(node_id, key_code) },

const struct gpio_keys_config kcfg[] = {
    DT_FOREACH_CHILD(DT_NODELABEL(buttons), INIT_KEY_CONFIG) /* note: requires `buttons: buttons {` in DT */
};

const size_t kcfg_cnt = ARRAY_SIZE(kcfg);
```

I haven't tested, but I think it should work. Note that `gpio-keys` compatible doesn't support `key-code` property, so you'll need to create your compatible.

ck-telecom, 2021-09-29:
hum...maybe this is useful when someone want beautiful code for numbers of keys

but why we use
.compatible = "gpio_keys"

with nothing to help and confusion.

spekary, 2022-04-22:
It seems strange to me that the gpio interrupt handler has no way of reliably querying what pin caused the interrupt, nor what edge (rising or falling) triggered it, and therefore cannot query the value of the pin. The only way to do it is to create a separate handler for each pin, rather than sharing the handlers. The fact that the handler sends the pin mask of all the pins that might trigger the handler, to the handler, is even stranger since that info is not that useful.

henrikbrixandersen, 2022-04-26:
> It seems strange to me that the gpio interrupt handler has no way of reliably querying what pin caused the interrupt, nor what edge (rising or falling) triggered it, and therefore cannot query the [...]

This is a typical limitation of the underlying hardware.

spekary, 2022-04-26:
Seems like a limitation of Zephyr. Zephyr already associates a callback and data structure with a pin, and then lets you configure how the callback is triggered, as in which edge triggers it. Seems like it would not be too difficult to save some of this information in the callback data structure and return it to the handler, since the callback data and the pins are already returned. Recording the edge that triggered might require driver support. There is lots of hardware that supports knowing which edge is triggering. Zephyr simply doesn't support exposing that to the app if its available.

ck-telecom, 2022-04-27:
> Seems like it would not be too difficult to save some of this information in the callback data structure and return it to the handler, since the callback data and the pins are already returned.

Could Zephyr try to save context and pass each pin handler?

spekary, 2022-04-27:
That should be really easy with a user data field in the callback data. Maybe a void*? Just annotate that the data can't be on the stack.

henrikbrixandersen, 2022-04-27:
> That should be really easy with a user data field in the callback data. Maybe a void*? Just annotate that the data can't be on the stack.

This can already be done using the existing GPIO API. Please see the exception mentioned in https://docs.zephyrproject.org/latest/develop/api/design_guidelines.html#using-callbacks.

henrikbrixandersen, 2023-03-23:
Closing this as we now have a generic GPIO keys input driver (see `drivers/input/input_gpio_keys.c`).

closed 2023-03-23 by henrikbrixandersen