zephyrproject-rtos/zephyr #38943: west: update hal_espressif failure

WilliamGFish, 2021-09-28:
**Describe the bug**
Error with west update function, fails to update hal_espressif

**To Reproduce**
Steps to reproduce the behavior:
1. west update, produces fatal: remote error: upload-pack: not our ref 09549c03111a2b330371d361d56227a457af66c5

> === updating hal_espressif (modules/hal/espressif):
> --- hal_espressif: fetching, need revision 0d30edf01a41b2c12e6715ff909f81421e96127e
> remote: Enumerating objects: 4545, done.
> remote: Counting objects: 100% (1219/1219), done.
> remote: Compressing objects: 100% (3/3), done.
> remote: Total 4545 (delta 1217), reused 1216 (delta 1216), pack-reused 3326
> Receiving objects: 100% (4545/4545), 1.04 MiB | 5.28 MiB/s, done.
> Resolving deltas: 100% (3081/3081), completed with 407 local objects.
> From https://github.com/zephyrproject-rtos/hal_espressif
>  * [new branch]          bugfix/espc32c3_eco0 -> refs/west/bugfix/espc32c3_eco0
>  * [new branch]          bugfix/warning_redefinition_esp32 -> refs/west/bugfix/warning_redefinition_esp32
>  * [new branch]          feature/esp32c3_ble  -> refs/west/feature/esp32c3_ble
>  * [new branch]          feature/esp32c3_boot_zephyr_sdk -> refs/west/feature/esp32c3_boot_zephyr_sdk
>  * [new branch]          feature/esp32c3_intc -> refs/west/feature/esp32c3_intc
>  * [new branch]          feature/esp32c3_support -> refs/west/feature/esp32c3_support
>  * [new branch]          feature/esp32c3_wifi -> refs/west/feature/esp32c3_wifi
>  * [new branch]          feature/gpio_esp32c3 -> refs/west/feature/gpio_esp32c3
>  * [new branch]          hal_v4.2             -> refs/west/hal_v4.2
>  * [new branch]          hal_v4.3             -> refs/west/hal_v4.3
>  * [new branch]          zephyr               -> refs/west/zephyr
>  * [new branch]          zephyr_v4.3          -> refs/west/zephyr_v4.3
> Fetching submodule components/esptool_py/esptool
> From https://github.com/espressif/esptool
>    1db7a23..e455e7d  master     -> origin/master
> fatal: remote error: upload-pack: not our ref 09549c03111a2b330371d361d56227a457af66c5
> Errors during submodule fetch:
>         components/esptool_py/esptool

**Expected behavior**
Update hal from git repository

**Impact**
Unknown...

carlescufi, 2021-09-28:
I´ve seen this multiple times already too. I think it's something to do with the submodules. Can you try running it again?

cfriedt, 2021-09-28:
@WilliamGFish - have you rebased /zephyr latetly? What commit sha are you on?

sylvioalves, 2021-09-28:
> I´ve seen this multiple times already too. I think it's something to do with the submodules. Can you try running it again?

You have seen this issue happening during `hal_espressif` update? I haven't yet. Perhaps we need to work this out to avoid it.

WilliamGFish, 2021-09-29:
When run west again the error doesn't appear but I'm not sure if there is an underlying issue.

I am always running the latest version by merging changes and run a west update at least once a day to assess updates.

Billy..

sylvioalves, 2021-09-29:
@WilliamGFish What OS?

nashif, 2021-09-29:
based on previous experience, this is most likely a network issue or a local issue, either way, reopen if you see this problem again.

closed 2021-09-29 by nashif