zephyrproject-rtos/zephyr #38959: ITE RISCV I2C driver returning positive values for error instead of negative values

Asthachavda, 2021-09-29:
**Describe the bug**
While tried to bind the i2c driver, wrote the 1-byte data in particular address (using i2c_write API), received positive number as an error instead of negative error code.   

**To Reproduce**
Steps to reproduce the behavior:
1. Use any I2C API like i2c_write on RISCV32 based ITE 8XXX2 EVB board using ITE's I2C driver.
2. Configure as I2C master
3. Don't connect any I2C slave device
4. Try to read or write and simulate any I2C error

**Expected behavior**
It must be giving Negative number for error or zero for success as per the Standard. 
(https://docs.zephyrproject.org/apidoc/latest/group__i2c__interface.html#ga2cc5f49493dce89e128ccbfa9d6149a0)

**Impact**
EC app. code fail to understand the error, because of wrong return code.

**Logs and console output**
***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v2.6.0-804-gb7b7709eecb0  (delayed boot 1ms) ***
Failed to write 8

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 18.04.5)
 - Toolchain: Zephyr SDK 12.04
 - Board used: ITE IT8XXX2 (RISC-V 32)

yingmingx, 2021-09-30:
Hi, in reproduce step 3, you haven't connect any slave device. I'm confused on it. 
Because when we use i2c, we should physically connect the device with the host controller. Then the host will send device address bitor R/W bit and hoping to get the ACK. 
If we haven't connect the slave device physically, we cannot get the right response.
Could you please check you hardware connection？

Asthachavda, 2021-09-30:
Hi, We used aardvark I2C host adapter as a master.

NagarajuGundimi, 2021-09-30:
Issue is NOT about whether I2C driver is working or not, Its about why ITE I2C driver returning +ve values for error instead of -ve values. This should have been caught in code reviews itself. No need to have any test setup and test case to reproduce it.

PR #39078 (2021-10-04): ITE: drivers/i2c: returning negative values for error
Fixes: #38959

Currently, the I2C driver returns I2C status register value as error
code when error happen. This PR fixes returning system number and
the return values is negative for error.

---
This change is [[image]](https://reviewable.io/reviews/zephyrproject-rtos/zephyr/39078)
commit 15106009 (2021-10-04): ITE: drivers/i2c: returning negative values for error

Fixes: #38959

Currently, the I2C driver returns I2C status register value as error
code when error happen. This PR fixes returning system number and
the return values is negative for error.
commit 8dacd0f8 (2021-10-05): ITE: drivers/i2c: returning negative values for error

Fixes: #38959

Currently, the I2C driver returns I2C status register value as error
code when error happen. This PR fixes returning system number and
the return values is negative for error.
closed 2021-10-04 by nashif