zephyrproject-rtos/zephyr #38966: Please add STM32F412VX

tfir1981, 2021-09-29:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
There are many kinds of STM32F4 microcontrollers. I am using the stm32f412ve in my project but this type is not maintained yet. In my opinion it is only the Kconfig, which is not maintained.

**Describe why you are asking for this support?**
I think this type is also widespread and should be mainline maintained. This support in hal_stm32 is given.

For me i fixed the issue temporarly.

**Additional context**
In Directory `soc/arm/st_stm32/stm32f4` i did:

add to _Kconfig.soc_:

```
config SOC_STM32F412VX
	bool "STM32F412VX"
```

add new file _Kconfig.defconfig.stm32f412vx_:
```
# ST STM32F412VX MCU configuration options
# SPDX-License-Identifier: Apache-2.0

if SOC_STM32F412VX

config SOC
	default "stm32f412vx"

config NUM_IRQS
	default 97

endif # SOC_STM32F412VX
```

FRASTM, 2021-09-29:
Maybe defining a Kconfig.defconfig.stm32f412**xe** is more pertinent to distinguish STM32F412xE or STM32F412xG  devices
(x is about pin count)

tfir1981, 2021-09-29:
I chose vx because of the implementations in hal_stm32 similar.

FRASTM, 2021-09-29:
Right, ../modules/hal/stm32/stm32cube/stm32f4xx/soc/stm32f412vx.h  exists
Please have a look at th PR https://github.com/zephyrproject-rtos/zephyr/pull/38986

Need a board to finalize (even if a dummy _nucleo_f412ve_ can build the samples/hello_world successfully)

tfir1981, 2022-05-31:
Hi, 
this is lost in zephyr 2.6, 2.6, 3 again.