zephyrproject-rtos/zephyr #38970: LWM2M Client Sample with DTLS enabled fail to connect

icatorze, 2021-09-29:
**Describe the bug**
I'm trying to run LwM2M client sample with DTLS enabled on a Nucleo_F207ZG board, but this error is reported on uart console:
 **`<err> net_lwm2m_engine: Failed to send packet, err 12`**

When debugging I get a -1 output over this fuction in 1048 line of lwm2m_engine.c file
```
static int lwm2m_send_message(struct lwm2m_message *msg){ 
...
    rc = send(msg->ctx->sock_fd, msg->cpkt.data, msg->cpkt.offset, 0);
    if (rc < 0){
...

```
Previously I could successful run this sample without DTLS.
I take a Linux PC to run Leshan Server as target. I modify prj.conf to aim this server and network configurations.

**To Reproduce**
Steps to reproduce the behavior:
`west build -b nucleo_f207zg -d build\lwm2m_clientdtls_f207 zephyr\samples\net\lwm2m_client -- -DCONF_FILE="prj.conf overlay-dtls.conf"`
`west flash -d build\lwm2m_clientdtls_f207`

**Expected behavior**
Board complete registration on Leshan Server.

**Impact**
Sample doesnt connect to lwm2m as expected.

**Logs and console output**
uart:~$ *** Booting Zephyr OS build v2.7.0-rc3-39-gffaff520eea4  ***
[00:00:01.574,000] <dbg> net_lwm2m_engine.lwm2m_engine_init: LWM2M engine socket receive thread started
[00:00:01.574,000] <dbg> net_lwm2m_obj_security.security_create: Create LWM2M security instance: 0
[00:00:01.574,000] <dbg> net_lwm2m_obj_server.server_create: Create LWM2M server instance: 0
[00:00:01.574,000] <dbg> net_lwm2m_obj_device.device_create: Create LWM2M device instance: 0
[00:00:01.574,000] <dbg> net_lwm2m_obj_firmware.firmware_create: Create LWM2M firmware instance: 0
[00:00:01.574,000] <inf> net_config: Initializing network
[00:00:01.574,000] <inf> net_config: Waiting interface 1 (0x20000e20) to be up...
[00:00:02.064,000] <inf> net_config: Interface 1 (0x20000e20) coming up
[00:00:02.065,000] <inf> net_config: IPv4 address: 192.168.1.199
[00:00:02.065,000] <inf> net_lwm2m_client_app: Run LWM2M client
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_set: path:0/0/2, value:0x2000da77, len:1
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_set: path:0/0/3, value:0x802b5dc, len:15
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_set: path:0/0/5, value:0x20000b9f, len:16
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_set: path:0/0/10, value:0x2000da76, len:2
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_set: path:1/0/0, value:0x2000da76, len:2
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_create_obj_inst: path:3303/0
[00:00:02.065,000] <dbg> net_ipso_temp_sensor.temp_sensor_create: Create IPSO Temperature Sensor instance: 0
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_create_obj_inst: path:3311/0
[00:00:02.065,000] <dbg> net_ipso_light_control.light_control_create: Create IPSO Light Control instance: 0
[00:00:02.065,000] <dbg> net_lwm2m_engine.lwm2m_engine_create_obj_inst: path:3340/0
[00:00:02.065,000] <dbg> net_ipso_timer.timer_create: Create IPSO Timer instance: 0
[00:00:02.065,000] <inf> net_lwm2m_rd_client: Start LWM2M Client: nucleo_f207zg
[00:00:02.574,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/0/1, buf:0x2000d1b7, buflen:1
[00:00:02.574,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/0/10, buf:0x2000d1e2, buflen:2
[00:00:02.574,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'nucleo_f207zg' with client lifetime 0
[00:00:02.574,000] <dbg> net_lwm2m_engine.lwm2m_parse_peerinfo: Parse url: coaps://192.168.1.142
[00:00:02.575,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:1/0/1, buf:0x2000d1a8, buflen:4
[00:00:02.575,000] <dbg> net_lwm2m_rd_client.sm_send_registration: registration sent [192.168.1.142]
**[00:00:02.576,000] <err> net_lwm2m_engine: Failed to send packet, err 12**
**[00:00:05.408,000] <err> net_lwm2m_engine: Failed to send packet, err 12**
[00:00:02.466,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/0/1, buf:0x2000d1b7, buflen:1
[00:00:02.466,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/0/10, buf:0x2000d1e2, buflen:2
[00:00:02.466,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'nucleo_f207zg' with client lifetime 0
[00:00:02.466,000] <dbg> net_lwm2m_engine.lwm2m_parse_peerinfo: Parse url: coaps://192.168.1.142
[00:00:02.466,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:1/0/1, buf:0x2000d1a8, buflen:4
[00:00:02.467,000] <dbg> net_lwm2m_rd_client.sm_send_registration: registration sent [192.168.1.142]
**[00:00:02.467,000] <err> net_lwm2m_engine: Failed to send packet, err 12**
**[00:00:05.453,000] <err> net_lwm2m_engine: Failed to send packet, err 12
[00:00:11.425,000] <err> net_lwm2m_engine: Failed to send packet, err 12
[00:00:23.369,000] <err> net_lwm2m_engine: Failed to send packet, err 12
[00:00:47.257,000] <err> net_lwm2m_engine: Failed to send packet, err 12**
[00:01:35.033,000] <wrn> net_lwm2m_rd_client: Registration Timeout

Wireshark doesn't registry any traffic.

**Environment (please complete the following information):**
 - OS: (Windows 10)
 - Toolchain (gnuarmemb (c:/nxp/MCUXpressoIDE_11.3.0_5222/ide/tools))

rlubos, 2021-09-30:
`err 12` indicates `ENOMEM` error, likely forwarded from mbed TLS. I'd try to increase `CONFIG_MBEDTLS_HEAP_SIZE` first (can be found in `overlay-dtls.conf`), perhaps this platform needs more. If this doesn't help, we'd need to get some mbedTLS logs to see where the problem is.

icatorze, 2021-09-30:
I increase `CONFIG_MBEDTLS_HEAP_SIZE` to 12000 and LWM2M sample over DTLS works now. @rlubos Thank you for your help.

closed 2021-09-30 by icatorze