zephyrproject-rtos/zephyr #38987: Unable to build ESP32 example code using west tool - zephyr

ashigupta, 2021-09-29:
Zephyr installation steps followed from : https://docs.zephyrproject.org/latest/getting_started/index.html
ESP32 setup guide followed: https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html 

I am using python virtual environment for this setup. 

I am able to install Python 3.9.0, Cmake and west tool. I have followed ESP32 setup guide (above link) to setup esp32 environment but unable to compile the example code.
Below are the issues i am facing:

1. On ESP32 setup guide, it is mentioned to set ESPRESSIF_TOOLCHAIN_PATH to "${HOME}/.espressif/tools/zephyr" however after  installing espressif toolchaing using command "west espressif install" as mentioned on same page, there is no zephyr folder created inside .espressif/tools/. Is this the correct path ? Is there any issue with installation ? Below is the screenshot of command and folder 
See Attachment 1
[image]

2.  While giving build command "west build -b esp32 samples/hello_world", i am getting toolchain error "Zephyr was unable to find the toolchain.  Is the environment misconfigured?"
I have already followed set environment variables command as mentioned in step 1 but still it is not setting though it says that 
 "Found toolchain: espressif (C:/Users/E0532110/.espressif/tools)"
[image]

 See Attachment 2

**To Reproduce**
Steps to reproduce the behavior:
1. Follow Zephyr installation steps
2. Follow ESP32 setup guide
3. Build code using command west build -b esp32 samples/hello_world

**Expected behavior**
Code should get compiled

**Impact**
Due to this issue i am not able to proceed on ESP32 development.

**Logs and console output**
Attached logs screenshots

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: espressif
 - Zephyr version: 2.7.0

**Additional context**
ESP32 build steps are different on forums and zephyr links. Can you share one document that can help to build ESP32 code on zephyr from scratch i.e. guiding what all environment variables needs to be setup at what installation stage.

sylvioalves, 2021-09-30:
@ashigupta This documentation was updated (and also the toolchain script) for > 2.7.0 release. It means that if you checked out tag 2.7.0 (-rcx) that indeed won't work. Anyway, can you try follow the documentation of 2.6.0 instead?  
https://docs.zephyrproject.org/2.6.0/boards/xtensa/esp32/doc/index.html  

If you checkout latest main, then the documented procedure will work.

ashigupta, 2021-10-01:
@sylvioalves after cleaning up all esp environment variables i have set earlier and re-installing cmake i am able to proceed for compilation. However now its giving me multiple definition error , below is the screenshot of the same 
[image]

is there any other configuration needed to compile hello world code for ESP32 ?

sylvioalves, 2021-10-01:
@ashigupta The PR that fixed this issue was merged a few hours ago. Please, pull latest main again.

ashigupta, 2021-10-04:
@sylvioalves  thanks for the update. I have again pulled main and now getting this compilation error for example code - Hello World
[image]

Is there any other fix available that needs to be pulled for this error ?

ashigupta, 2021-10-04:
@sylvioalves  thanks for the help here. After pulling Main again i am able to build hello world code for ESP32 and able to flash the same.

I have some followed up questions, it will be great if you can help for them too

1. How we can change the sdkconfig using west tool .
    > I was using idf.py menuconfig earlier while using esp32 setup with eclipse and freertos

2. Is there any IDE available on windows that supports ESP32 development using Zephyr ?

3. Can we compile ESP-IDF example code directly using west tool ? If yes is this procedure documented any where ?

sylvioalves, 2021-10-04:
Hi @ashigupta, find the answer below:

> 1. How we can change the sdkconfig using west tool .
>    > I was using idf.py menuconfig earlier while using esp32 setup with eclipse and freertos  

You can use `west build -t menuconfig`. For more details, check this [link](https://docs.zephyrproject.org/2.4.0/guides/kconfig/menuconfig.html).

> 2. Is there any IDE available on windows that supports ESP32 development using Zephyr ?  

You can use VSCode, however, I think Zephyr has no ready-to-use extension for that. I think there is some ongoing talk regarding this for a while: Please, check this [thread](https://github.com/zephyrproject-rtos/zephyr/issues/21119).

> 3. Can we compile ESP-IDF example code directly using west tool ? If yes is this procedure documented any where ? 

No, these are 2 different frameworks. Either you use `west`, which is a meta-tool for Zephyr project, or you work with esp-idf and use `idf.py` tool.

ashigupta, 2021-10-07:
@sylvioalves  Thanks for the information. Is there any example code available which is using Zephyr OS instead of FreeRTOS for ESP32  that i can refer ?
I found only two example code in Zephyr repository for ESP32 - 1. SPI RAM and 2. Wifi Station mode. Below is the snippet of the same 
[image]

closed 2021-11-02 by sylvioalves