zephyrproject-rtos/zephyr #38994: ARP: Replies are sent to multicast MAC address rather than senders MAC address.

rmelch, 2021-09-29:
ARP replies are being sent to IPv4mcast MAC addresses rather than the expected source MAC address. I have attached 2 pcap files:
  1. arp-reply-mcast.pcapng
  2. arp-reply-mcast-fixed.pcapng
  
The first shows the issue: 

*Ping 172.16.225.92 from 172.16.225.94*

`1	2021-09-30 11:04:31.281	TexasIns_1a:9b:18	ARP	Broadcast			Who has 172.16.225.92? Tell 172.16.225.94`
`2	2021-09-30 11:04:31.281	STMicroe_00:87:35	ARP	IPv4mcast_5c:b8:80			172.16.225.92 is at 00:80:e1:00:87:35`

**Note** the Destination is  `IPv4mcast_5c:b8:80` I believe it should be TexasIns_1a:9b:18 as seen in the 2nd pcap file.

*Ping 172.16.225.92 from 172.16.225.91*

`5	2021-09-30 11:24:31.816	MitacInt_9a:2e:8b	ARP	Broadcast			Who has 172.16.225.92? Tell 172.16.225.91`
`6	2021-09-30 11:24:31.816	STMicroe_00:87:35	ARP	MitacInt_9a:2e:8b			172.16.225.92 is at 00:80:e1:00:87:35`

**To Reproduce**
Steps to reproduce the behaviour:
1. Using Zephyr 2.6 
2. Using Socket Echo Server as example
3. west build -b nucleo_f429zi ../zephyr/samples/net/sockets/echo_server
4. Use wireshark to capture packets.
5. Ping the target from host to trigger an ARP request.

**Expected behavior**
APR Replies are sent to the senders MAC address, not the mcast address.

**Impact**
ARP only works when NIC is in promiscuous mode. 

**Logs and console output**
See attached

Possible Fix:

```
diff --git a/subsys/net/l2/ethernet/ethernet.c b/subsys/net/l2/ethernet/ethernet.c
index 7089d406c9..ca2647e471 100644
--- a/subsys/net/l2/ethernet/ethernet.c
+++ b/subsys/net/l2/ethernet/ethernet.c
@@ -494,8 +494,9 @@ static struct net_buf *ethernet_fill_header(struct ethernet_context *ctx,

hdr_vlan = (struct net_eth_vlan_hdr *)(hdr_frag->data);

-               if (!ethernet_fill_in_dst_on_ipv4_mcast(pkt, &hdr_vlan->dst) &&
-                   !ethernet_fill_in_dst_on_ipv6_mcast(pkt, &hdr_vlan->dst)) {
+               if (ptype == htons(NET_ETH_PTYPE_ARP) ||
+                       (!ethernet_fill_in_dst_on_ipv4_mcast(pkt, &hdr_vlan->dst) &&
+                   !ethernet_fill_in_dst_on_ipv6_mcast(pkt, &hdr_vlan->dst))) {
memcpy(&hdr_vlan->dst, net_pkt_lladdr_dst(pkt)->addr,
sizeof(struct net_eth_addr));
}
@@ -515,8 +516,9 @@ static struct net_buf *ethernet_fill_header(struct ethernet_context *ctx,
} else {
hdr = (struct net_eth_hdr *)(hdr_frag->data);

-               if (!ethernet_fill_in_dst_on_ipv4_mcast(pkt, &hdr->dst) &&
-                   !ethernet_fill_in_dst_on_ipv6_mcast(pkt, &hdr->dst)) {
+               if (ptype == htons(NET_ETH_PTYPE_ARP) ||
+                       (!ethernet_fill_in_dst_on_ipv4_mcast(pkt, &hdr->dst) &&
+                   !ethernet_fill_in_dst_on_ipv6_mcast(pkt, &hdr->dst))) {
memcpy(&hdr->dst, net_pkt_lladdr_dst(pkt)->addr,
sizeof(struct net_eth_addr));
}
```

rmelch, 2021-09-29:
Add pcap files.
[pcap-files.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/7255584/pcap-files.tar.gz)

jukkar, 2021-09-30:
Please send a PR with your proposal, it is difficult to review code from issue.

mrfuchs, 2021-10-06:
Anyone an idea why this expression is evaluating to false for @rmelch's data above?

https://github.com/zephyrproject-rtos/zephyr/blob/3ebe9c9ea122fb251a0574f0ef582ce9aedd34c3/subsys/net/l2/ethernet/ethernet.c#L544-L545

I just traced this with Wireshark with a patched and unpatched version of ethernet.c and it works in both cases (response is sent to senders MAC address).

rmelch, 2021-10-06:
Is it valid to look in an IP header for APR packets? It could be that NET_IPV4_HDR(pkt) is not valid/initialised.

rmelch, 2021-10-06:
Looking at this ARP Frame from wireshark:

`2	2021-09-30 11:04:31.281	STMicroe_00:87:35	ARP	IPv4mcast_5c:b8:80	172.16.225.92 is at 00:80:e1:00:87:35`

```
0000   01 00 5e 5c b8 80 00 80 e1 00 87 35 08 06 00 01
0010   08 00 06 04 00 02 00 80 e1 00 87 35 ac 10 e1 5c
                                           ^------- ARP sender IP addr
                                                 ^------- IP4 frame dest add.
0020   b8 80 4f 1a 9b 18 ac 10 e1 5e 00 00 00 00 00 00
0030   00 00 00 00 00 00 00 00 00 00 00 00

```
The ARP sender address is `ac 10 e1 5c` (172.16.225.92). If interpreted as an IPV4 frame
the destination address would be `e1 5c b8 80` (225.92.184.128) which is why the address
is interpreted as multi cast `IPv4mcast_5c:b8:80`

commit 59fcf8c0 (2021-10-07): net: ethernet: ARP addresses being filled with mcast addresses

Fixes #38994, ARP messages were being sent to IPvXmcast MAC addresses
rather than the expected source MAC address or the broadcast address.
commit e7df33e8 (2021-10-12): net: ethernet: ARP addresses being filled with mcast addresses

Fixes #38994, ARP messages were being sent to IPvXmcast MAC addresses
rather than the expected source MAC address or the broadcast address.