zephyrproject-rtos/zephyr #38996: The coap server is missing a leave multicast group method

AllemanMa, 2021-09-30:
**Describe the bug**
It is possible for the coap_server to join a multicast group, however it is not possible to leave a multicast group. There is no implementation of `NET_EVENT_IPV6_CMD_MADDR_DEL` in `ipv6_addr_event_handler` in `openthread.c`. 

Building the coap_server with Openthread support fails because of `net_ipv6_mld_join`. Replacing this method with `net_if_ipv6_maddr_add` and  `net_if_ipv6_maddr_join` seems to solve the problem. 

An issue related to the 2 points above has been reported here: https://github.com/zephyrproject-rtos/zephyr/issues/17692. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build the coap_server sample with Openthread
2. Build process fails
3. Change `net_ipv6_mld_join` to `net_if_ipv6_maddr_add` and `net_if_ipv6_maddr_join` to be able to build the sample.
4. There is no way of leaving a multicast group

**Expected behavior**
I expected that there were a method to leave a certain multicast group.

**Impact**
I am not able to leave a multicast group.

**Environment (please complete the following information):**
 - Toolchain (e.g Zephyr SDK, ...)
 - Zephyr: v2.6.99-ncs1-rc2

cfriedt, 2021-09-30:
@AllemanMa - are you able to reproduce this upstream against, e.g. v2.7.0-rc3?

AllemanMa, 2021-10-01:
> @AllemanMa - are you able to reproduce this upstream against, e.g. v2.7.0-rc3?

I am developing with Nordic Connect SDK which does not have such a recent release yet of Zephyr, that's why I am still on v2.6.99-ncs1-rc2. When I have a look at the Zephyr repository on the master branch, I can see that there is still no implementation for `NET_EVENT_IPV6_ADDR_DEL  ` in https://github.com/zephyrproject-rtos/zephyr/blob/e0f18128bfd23c99a060a5f931c1bcaa6afdab6a/subsys/net/l2/openthread/openthread.c#L117

Als no method like `rm_ipv6_maddr_from_ot` is provided in https://github.com/zephyrproject-rtos/zephyr/blob/e0f18128bfd23c99a060a5f931c1bcaa6afdab6a/subsys/net/l2/openthread/openthread_utils.c#L336 .

In the coap_server sample, there is no implementation for `leave_coap_multicast_group` available either.

AllemanMa, 2021-10-04:
It seems like `join_coap_multicast_group()` of the coap server sample can be replaced by `otIp6SubscribeMulticastAddress` and leaving a multicast address group by `otIp6UnsubscribeMulticastAddress`. These methods provide the functionality I was expecting.

closed 2021-10-05 by AllemanMa