zephyrproject-rtos/zephyr #39009: Nordic PWM causing lock up due to infinte loop

lairdjm, 2021-09-30:
**Describe the bug**
We are using a Nordic PWM for control of 2 LEDs, in very random cases we get a module lock up which is caused by https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/pwm/pwm_nrfx.c#L239 in which case it keeps running the loop. It seems we can get the issue to occur within a minute by flicking the LED states to be PWM to off and back again.

**To Reproduce**
PWM led which changes state from PWM to off and back again and rapidly change it

**Expected behavior**
The code to work without a lock up

**Impact**
As this kills the code right it it's tracks, show stopper.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: Using nRF connect SDK 1.6.1 so zephyr 2.6

lairdjm, 2021-09-30:
Just to add: Using an nRF5240 revision C silicon, two PWM instances (one per LED), for debugging I've added code to exit from the loop when it's sat there for a while and sometimes it hits LED0, sometimes LED1 and something both LEDs, seems random. When this has happened, I've checked the PWM registers and when this happens EVENTS_STOPPED is 1, EVENTS_SEQSTARTED1 is 1, EVENTS_SEQEND0 is 1, EVENTS_SEQEND1 is 1, EVENTS_PWMPERIODEND is 1, EVENTS_LOOPSDONE is 1 (EVENTS_SEQSTARTED0 is 0)

anangl, 2021-10-04:
> It seems we can get the issue to occur within a minute by flicking the LED states to be PWM to off and back again.
> 
> **To Reproduce**
> PWM led which changes state from PWM to off and back again and rapidly change it

I tried to reproduce this but so far I didn't manage to.
Could you at least provide some code snippet(s) that would show how this switching off and on again is realized (and ideally a simple app on which the issue is reproducible)? I assume that by changing the state to off you mean setting the pulse width for a given LED to 0, right?
Could you specify the PWM periods you are using for those LEDs and how fast is the switching performed, also whether this is done from one context only?

> When this has happened, I've checked the PWM registers and when this happens EVENTS_STOPPED is 1, 

That's really strange. I can't quite imagine how this is possible, since the loop you pointed out is:
https://github.com/zephyrproject-rtos/zephyr/blob/ab6c090d74f91d7904bc71718e1e7642ee8c603a/drivers/pwm/pwm_nrfx.c#L239-L240
and `nrfx_pwm_is_stopped()` returns `true` [when the STOPPED event is set](https://github.com/zephyrproject-rtos/hal_nordic/blob/a6e5299041f152da5ae0ab17b2e44e088bb96d6d/nrfx/drivers/src/nrfx_pwm.c#L430-L435):
```
    if (nrf_pwm_event_check(p_instance->p_registers, NRF_PWM_EVENT_STOPPED))
    {
        p_cb->state = NRFX_DRV_STATE_INITIALIZED;
        NRFX_LOG_INFO("Disabled.");
        ret_val = true;
    }
```

carlescufi, 2021-10-06:
@lairdjm could you please update this issue with more info as requested by @anangl?

lairdjm, 2021-10-06:
Not this week due to project deadlines, will see if I get time to create a minimal application that replicates it next week

JeremyBCD, 2021-10-07:
@anangl @carlescufi I think that I have run into this same issue, and I've created a small project that reproduces it: https://github.com/JeremyBCD/zephyr/blob/pwm-lock-up/samples/hello_world/src/main.c

I've modified `hello_world` to bring in the PWM functionality I'm working on (buzzer stuff in my case).

anangl, 2021-10-07:
@JeremyBCD Thank you for providing this application. I was able to reproduce the problem and I provided a fix for it in #39239. Actually, the call in line [76](https://github.com/JeremyBCD/zephyr/blob/db2b84f0e036a314f7dd8e37c6b76a6be34fba30/samples/hello_world/src/main.c#L76):
```
pwm_pin_set_usec(pwm_dev, PWM_CHANNEL, 0, 0, 0);
```
is a bit problematic, as for zero length period some PWM drivers will return `-EINVAL`. In the PR mentioned above, I fixed the pwm_nrfx driver to be more flexible in this regard, basically to not cause an undefined behavior of the PWM peripheral, but for portable code this call should be rather:
```
pwm_pin_set_usec(pwm_dev, PWM_CHANNEL, note_period, 0, 0);
```