zephyrproject-rtos/zephyr #39025: Bluetooth: Periodic Advertising, Filter Accept List, Resolving list related variable name abbreviations

cvinayak, 2021-10-01:
**Is your enhancement proposal related to a problem? Please describe.**
Periodic Advertising, Periodic Advertising List, Filter Accept List, and Resolving List use abbreviations `per_adv`, `pal`, `fal` and `rl` respectively, which are "It is hard to figure out what is it until you are in the topic." 

- Originally, `pl` was used for Periodic Advertising List as `pal` would overlapped with 802.11 PAL in BT Spec
- `pa` is popular for Power Amplifier
- `PER` is abbreviation for Peripheral in the Bluetooth Test Specification
- `per` is a word in itself having different meaning, like in "as per" something

Refer to discussions here:
- https://github.com/zephyrproject-rtos/zephyr/pull/38338#pullrequestreview-766831451
- https://github.com/zephyrproject-rtos/zephyr/pull/38338#discussion_r719172107
- https://github.com/zephyrproject-rtos/zephyr/pull/38338#discussion_r718617791

**Describe the solution you'd like**
- "So if not pal and not per_adv_list, I guess we are just left with pa_list or periodic_adv_list (or even periodic_advertiser_list)." 

Baseline the variable abbreviations in this issue, and take actions to refactor implementation to reflect the decided namings.

**Describe alternatives you've considered**
Keep the `per_adv`, `pal`, `fal` and `rl` until a consensus is reached on an adopted rules for variable naming

**Additional context**
None for now.

carlescufi, 2021-10-01:
I see no real reason not to use `pal`, `fal` and `rl`. Those are the literal abbreviations of the spec terms, and this is internal code, not public APIs. My vote goes to staying with those.

ppryga, 2021-10-01:
I've got different opinion on that. 

Even the code is internal it should be as readable as possible. That would not intimidate new developers that want to contribute to the implementation. 
The abbreviations are not used in BT Core Spec. @Thalley can confirm that, because he has verified it.
What more there is a number of abbreviations used in the Bluetooth stack. These are ok for people that are already experienced in development of the stack and read the code related with periodic advertiser list, filter accept list, resolving list. If one hadn't been familiar with the features of the protocol and hadn't done anything in that areas of the stack, then it would be hard to guess what the abbreviations mean.

In my humble opinion, we should avoid making steeper already steep learning curve of the stack implementation.

Thalley, 2021-10-01:
Furthermore, `pal` (PAL) was the spec acronym for the Protocol Adaptation Layer from Core 5.2 and earlier (remove in 5.3). 

I personally think `fal` and `pal` are fine to use, but I also agree with @ppryga that they may be a bit tricky.

jori-nordic, 2023-02-16:
Closing as won't do since it's been more than a year and people are mostly okay with keeping the current abbrevs.

I opened https://github.com/zephyrproject-rtos/zephyr/issues/54893 instead, we'll try to have a first version in the next few months.

closed 2023-02-16 by jori-nordic