zephyrproject-rtos/zephyr #39036: Multicast packet forwarding not working for the coap_server sample and Openthread

AllemanMa, 2021-10-01:
**Describe the bug**
Multicast packet forwarding in combination with Openthread is not working. I have a setup of 4 devices. 

1 Leader
2 Router
3 Child_L (child of leader)
4 Child_R (child of Router)

   (Leader)----------(Router)
         |                          |
         |                          |
  (Child_L )            (Child_R)

If the Leader sends a multicast packet, the Router and Child_L will receive the multicast packet. However Child_R won't. It seems like the packet is not forwarded over the Router towards its child. 

**To Reproduce**
Steps to reproduce the behavior:
1. build coap_server sample with Openthread support
2. Send a coap multicast message from the leader, for instance to mcast address: [ff03::1]
3. Watch all the devices receiving the packet except for Child_R

**Expected behavior**
I expect all devices to receive the multicast message.

**Impact**
I am not able to address my whole Openthread mesh network using one single multicast packet.

**Environment (please complete the following information):**
 - nrf: v1.7.0-rc2
 - zephyr: v2.6.99-ncs1-rc2

AllemanMa, 2021-10-01:
Everytime a multicast packet is transmitted, the following is printed out at Child_R: 
```
[00:01:32.471,221] <inf> net_openthread: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:68, seqnum:219, type:Data, src:0xa000, dst:0xffff, sec:yes, ackreq:no
[00:01:32.496,459] <inf> net_openthread: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:68, seqnum:221, type:Data, src:0xa000, dst:0xffff, sec:yes, ackreq:no
```

AllemanMa, 2021-10-05:
Using a packet sniffer it has become clear that the parent does not retransmit the multicast packet towards its children. Could this be an issue with MPL?

carlescufi, 2021-10-06:
As per the Discord thread below, this seems the fault of the CoAP client setting the hop limit to 1:
https://discord.com/channels/720317445772017664/884102400149966930/894958468799287419

Please reopen if this is not the case and this is indeed an issue with Zephyr.