zephyrproject-rtos/zephyr #39037: samples/net/civetweb/websocket_server fails to build with CONFIG_NEWLIB_LIBC

icsys-omh, 2021-10-01:
**Describe the bug**
The websocket_server sample application fails to build with `conflicting types` errors on pthread-related types when compiled with CONFIG_NEWLIB_LIBC. 

I originally discovered this problem when working on a custom C++ application, where newlib was used to get some of the C++ standard library. However, the issue can be readily reproduced using the websocket server sample, as long as the libc_extension.c is removed from CMakeLists.txt. 

**To Reproduce**
Steps to reproduce the behavior:
1. Modify the websocket server sample to enable CONFIG_NEWLIB_LIBC, and remove the libc_extension.c file:
```
diff --git a/samples/net/civetweb/websocket_server/CMakeLists.txt b/samples/net/civetweb/websocket_server/CMakeLists.txt
index 022e360762..0cf01f8c90 100644
--- a/samples/net/civetweb/websocket_server/CMakeLists.txt
+++ b/samples/net/civetweb/websocket_server/CMakeLists.txt
@@ -23,7 +23,7 @@ project(civetweb_websocket_server)
 
 target_sources(app PRIVATE
                ${src_dir}/main.c
-               ${common_src_dir}/libc_extensions.c
                ${src_dir}/http_server_handlers.c
                ${src_dir}/websocket_server_handlers.c)
 
diff --git a/samples/net/civetweb/websocket_server/prj.conf b/samples/net/civetweb/websocket_server/prj.conf
index b582bdd921..771d45164a 100644
--- a/samples/net/civetweb/websocket_server/prj.conf
+++ b/samples/net/civetweb/websocket_server/prj.conf
@@ -1,6 +1,8 @@
 # General config
 CONFIG_CIVETWEB=y
 
+CONFIG_NEWLIB_LIBC=y
+
 # pthreads
 CONFIG_POSIX_API=y
 CONFIG_PTHREAD_IPC=y
```
2. Try to build the sample application with west:
`west build samples/net/civetweb/websocket_server/ -b nucleo_f767zi --build-dir build/`

**Expected behavior**
The build should succeed.

**Impact**
Our application uses code that relies on the C++ standard libary (std::string, std::chrono), so disabling newlib is hence not really an option. This hence means that Zephyr's Civetweb implementation is unusable.

**Logs and console output**
The full build log is this on Ubuntu 20.04 with Zephyr SDK 0.13.1:
```
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/username/Projects/subseadisplay/zephyr/cmake/app/boilerplate.cmake
CMake Deprecation Warning at /home/username/Projects/subseadisplay/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
  The OLD behavior for policy CMP0079 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  /home/username/Projects/subseadisplay/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/username/Projects/subseadisplay/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:21 (find_package)

-- Application: /home/username/Projects/subseadisplay/zephyr/samples/net/civetweb/websocket_server
-- Zephyr version: 2.7.0-rc1 (/home/username/Projects/subseadisplay/zephyr), build: v2.7.0-rc1-151-g81595ed4db20
-- Found Python3: /home/username/zephyr-venv/bin/python3 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: nucleo_f767zi
-- Cache files will be written to: /home/username/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.13.1 (/opt/zephyr-sdk-0.13.1)
[... 233 lines cut ...]
In file included from /home/username/Projects/subseadisplay/modules/lib/civetweb/src/civetweb.c:168:
/home/username/Projects/subseadisplay/zephyr/samples/net/civetweb/websocket_server/../common/include/libc_extensions.h:26:5: error: expected ')' before '+' token
   26 | int iscntrl(int c);
      |     ^~~~~~~
/home/username/Projects/subseadisplay/zephyr/samples/net/civetweb/websocket_server/../common/include/libc_extensions.h:26:5: error: expected ')' before '[' token
   26 | int iscntrl(int c);
      |     ^~~~~~~
/home/username/Projects/subseadisplay/zephyr/samples/net/civetweb/websocket_server/../common/include/libc_extensions.h:26:5: error: expected ')' before '&' token
   26 | int iscntrl(int c);
      |     ^~~~~~~
In file included from /home/username/Projects/subseadisplay/modules/lib/civetweb/src/civetweb.c:172:
/home/username/Projects/subseadisplay/zephyr/samples/net/civetweb/websocket_server/../common/include/libc_extensions.h:42:5: error: expected identifier or '(' before 'int'
   42 | int ferror(FILE *stream);
      |     ^~~~~~
/home/username/Projects/subseadisplay/zephyr/samples/net/civetweb/websocket_server/../common/include/libc_extensions.h:42:5: error: expected ')' before '(' token
   42 | int ferror(FILE *stream);
      |     ^~~~~~
[161/199] Building C object modules/stm32/stm32cube/CMakeFiles/..__...stm32__stm32cube.dir/stm32f7xx/drivers/src/stm32f7xx_ll_utils.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /snap/bin/cmake --build /home/username/Projects/subseadisplay/zephyr/build
```

**Environment (please complete the following information):**
The build log above is from Ubuntu 20.04 with Zephyr SDK 0.13.1, but the same issue appears on Windows using GNU ARM Embedded 10-2020-q4-major.
 - Zephyr commit 81595ed4db

stephanosio, 2021-10-01:
This is a known issue and a very unfortunate limitation imposed by the status quo.

From #16683:

> Using Newlib is not an option due to heavy conflicts with the POSIX layer (see #13787).

We will eventually address this issue by properly porting the newlib to the Zephyr (see https://github.com/zephyrproject-rtos/sdk-ng/issues/350).

BenjaminDeuter, 2021-12-16:
At [Geek in Business GmbH](https://geekinbusiness.de/) we are interested in using Civetweb on top of newlibc. [Thanks to some posts from @stephanosio about how defining _GNU_SOURCE affects newlibc](https://github.com/search?q=org%3Azephyrproject-rtos+_GNU_SOURCE&type=issues) , I was able to get the http_server and websocket_server sample to run successfully with only a few lines of patches. I am posting my work here in case other people may find it useful (tested using the GNU Arm Embedded toolchain on Windows):
https://github.com/BenjaminDeuter/zephyr/tree/poc_civetweb_newlibport
https://github.com/BenjaminDeuter/civetweb/tree/poc_civetweb_newlibport

I am aware that this fork does not properly integrate newlibc with Zephyr's POSIX subsystem but instead works around the [contradictions which were outlined above by @icsys-omh](https://github.com/zephyrproject-rtos/zephyr/issues/39037#issue-1013300700) by turning off functionality in newlibc.

@pfalcon
For reasons of maintainability, we are generally interested in getting Civetweb supported on top of newlibc in upstream Zephyr. As you stated that this requires on the order of "50-100 incremental changes" (https://github.com/zephyrproject-rtos/zephyr/pull/16009), it would be great if you could offer suggestions on where to start.

I assume Civetweb functions need to be ported and tested one after another. For example, I have seen that you were initally sceptical regarding the use of #include_next ( https://github.com/zephyrproject-rtos/zephyr/issues/24730 , https://github.com/zephyrproject-rtos/zephyr/issues/18892 ), but nevertheless https://github.com/zephyrproject-rtos/zephyr/pull/27939 was merged. As Civetweb makes heavy use of POSIX's pthread API, should #include_next be introduced similarly inside include/posix/pthread.h so that newlibc's pthread will be used instead of the Zephyr POSIX subsystem's pthread implementation?

gmarull, 2022-08-11:
civetweb is no longer part of Zephyr, closing

closed 2022-08-11 by gmarull