zephyrproject-rtos/zephyr #39061: [backport v2.7-branch] backport of #38960 failed

zephyrbot, 2021-10-01:
This issue was created automatically because the backport of #38960 to `v2.7-branch` failed.

The backport to `v2.7-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.7-branch v2.7-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.7-branch
# Create a new branch
git switch --create backport-38960-to-v2.7-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick 8b8a3bbf713264e926d358bde7b606d4148d1c58
# Push it to GitHub
git push --set-upstream origin backport-38960-to-v2.7-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.7-branch
```
Then, create a pull request where the `base` branch is `v2.7-branch` and the `compare`/`head` branch is `backport-38960-to-v2.7-branch`.

jakub-uC, 2021-10-14:
@cfriedt :
Sure

cfriedt, 2021-12-18:
Merged Oct 19

closed 2021-12-18 by cfriedt