zephyrproject-rtos/zephyr #39064: Provide different SSH credentials for git repositories in west manifest files.

blake-bit, 2021-10-01:
**Is your enhancement proposal related to a problem? Please describe.**
Currently when `west` accesses remote git repositories it cannot be provided with individual ssh keys for a specific repo. This means that if a `west.yml` manifest file has different private repositories (all on the same server, e.g. Github) which each require unique SSH keys, each key cannot be supplied to the underlying `git` calls to have the proper credentials to clone, fetch, checkout, etc. because west is only invoked once to access all given repos, e.g. `west init`.

On the command line can be worked around manually by temporarily invoking a new `ssh-agent` with a given unique key, and then running `git`; the ssh-agent will then terminate at the end of the script/session and the keys are not stored permanently in any configuration.
`ssh-agent bash -c 'ssh-add /somewhere/yourkey; git clone git@github.com:user/project.git'`

**Describe the solution you'd like**
It would be great to be able to provide a keyfile, or other method, to allow secure access to a given repository as a data member in the manifest file along with the other repo information.

**Describe alternatives you've considered**
This isn't an issue for our developers on their local builds as their credentials generally allow access to all of the repos we need to clone. 
However we have build systems with more restrictive credentials (e.g. Github Deploy Keys unique to each repo) where we have to invoke a new `ssh-agent` with the correct key each time before executing a `git` command. This means that before we invoke something like `west init` for a clean build, we manually clone and checkout the repos based on the information in the `west.yml` file before `west init` is allowed to run. When `west` sees the directories are already populated with the correct files from the git checkout, it quietly moves on and things function normally. However, this is a very cumbersome work around.

henrikbrixandersen, 2022-11-02:
Configuration of SSH+Git credentials needs to happen in SSH, i.e. via ~/.ssh/config. West will invoke Git, which will invoke SSH - and SSH will use the credentials provided via ~/.ssh/config.

henrikbrixandersen, 2022-11-02:
CC: @mbolivar-nordic